NanoVNA-Saver/nanovna-saver #7: Add more sophisticated calibration

RolandWa, 2019-09-02:
Dear Sirs,

Could be possible for test purposes add the more sophisticated method of calibration to the python script - form more points.

https://github.com/Ttl/vna2/blob/master/software/twoport/calibrate.py

More documentation at 
https://scikit-rf.readthedocs.io/en/latest/

Best Regards
Roland

mihtjel, 2019-09-02:
Hi Roland,
thanks for the links. I'll definitely take a look - it's a feature I want, but I haven't yet researched how to actually implement it.

/Rune

mihtjel, 2019-09-03:
I have implemented 1-port in-app calibration in a separate branch. It uses data from a sweep (done in the app, or loaded from file) for each of a short, open and load. Corrections seem to work, but I have only scratched the surface in testing it.

mihtjel, 2019-09-16:
1.5 port calibration using imperfect standards has been implemented. :-)

closed 2019-09-16 by mihtjel