NanoVNA-Saver/nanovna-saver #11: The number of digits is probably more than you could ever hope for with this VNA.

frohro, 2019-09-09:
I suspect the accuracy of S_11 and S_21 are fairly predictable, based on the specs published on the NanoVNA.  You could find the approximate delta of the calculated values for the markers using total differentials from Calculus, or just vary the S_11 or S_21 by the amount, and see the differences you calculate.  This could then be used to determine the digits.  If you wanted to be really cool, you could put the expected margin for error in as well.  :-)
Rob

mihtjel, 2019-09-09:
The software is set up for 3 decimals, which seems reasonable. Your screenshot has an impedance value that looks like something wasn't quite properly calibrated, which would account for the large number of digits?

mihtjel, 2019-09-09:
But if you have some good references on how to determine margin of error, I'm all ears :-)

frohro, 2019-09-09:
Hi Rune,
When you get an |S_11| > 1, I just wouldn't print a VSWR or impedance.  VSWR assumes there is only one source and the line and components along it are passive.  When |S_11| is slightly greater than 1, then the bogus VSWR (and bogus impedances with negative real parts) occur.  You could say |S_11| is greater than one; check your calibration.  I did a calibration using your code before the screenshots above.  I was measuring a length of line with an open circuit on it.  That is a rather tough test for a VNA, because small errors can be seen pretty easily.  If the line was perfectly lossless, the |S_11| would be identically 1, and any error would be pretty easy to spot.  With a lossy line the loss goes up as sqrt( frequency ) and so it should spiral in on the Smith chart as frequency increases.  Generally impedances near Z_0 (50 ohms in our case) are most accurately measured.  If you look at a Smith chart, you will see that changing S_11 just a very small amount when it is near 1 + j 0, will change the impedance immensely.  You could say |Z| > some number, but having three digits after the decimal is meaningless, and misleading.  Similarly when |S_11| is close to 1, the VSWR has a singularity (goes toward infinity), and if |S_11| is slightly greater than 1, it goes toward negative infinity, which couldn't be any further from + infinity than you can get!  Perhaps in those cases it would be safe to just say "VSWR is very high!" or something like that.

mihtjel, 2019-11-06:
I think the decimals have generally been reduced to more or less sensible amounts in all places, so I'm closing this :-) Thank you for your input!

closed 2019-11-06 by mihtjel