NanoVNA-Saver/nanovna-saver #16: Show bands in VSWRChart ( HAM + CB )

gaionim, 2019-09-18:
I like see if swr is near ham band.

I've some proof of concept , see image.

I can cleanup the code and try to do a Pull Request, but you have to integrate better or support me before pull.
( es.    select background color in config )

[image]

mihtjel, 2019-09-18:
Hi,
thank you for the suggestion, and the proof of concept. It looks very interesting!

I'm not sure what integration/support you are looking for from me, though?

gaionim, 2019-09-19:
Hi, I'll cleanup my code and I'll show you soon ( I hope )
Basically integration with the configuration because I'm new to QT.

73 de IU2HDS

gaionim, 2019-09-20:
see pull request 17

mihtjel, 2019-09-20:
Thank you for the pull request! I'll definitely look at implementing this functionality, but I don't think I'll be accepting the code directly, as I think I'd want the ability to select which bands get displayed, for example (and also to keep the code in English ;-)

I don't know that it will make the next release, but probably the one after that!

Thank you very much for your effort!

gaionim, 2019-09-20:
Ok, I'll wait ;-) 

73

closed 2019-09-20 by gaionim