NanoVNA-Saver/nanovna-saver #18: relax pyqt5 restriction

ZeroChaos-, 2019-09-20:
setup.py restricts to a very specific version of pyqt5 but when I test with 5.13.0 it works fine, perhaps it should be >= instead of ==?

https://github.com/mihtjel/nanovna-saver/blob/master/setup.py#L53

mihtjel, 2019-09-20:
Hi,
Using a newer version broke building using pyinstaller, so for now, it's set to a known working version in order to build working executables.

If 5.13.1 turns out to have fixed it, I'll change it again.

ZeroChaos-, 2019-09-23:
I can say nothing more than "it works for me".  It's a simple enough patch to carry in gentoo, so adjust it when it suits you, I'm all set.  thanks for the great software, I'm excited to use it more.

barjac, 2019-10-17:
I too hit this and patched it to be >= as my package for Mageia 7 would not install.
All working fine now 'out of the box' with my unmodified nanovna.
Now to read the docs!
Thanks for a great project and for supporting Linux :)

mihtjel, 2019-11-06:
Done for 0.1.5.

closed 2019-11-06 by mihtjel