NanoVNA-Saver/nanovna-saver #19: Add marker for TDR

gaionim, 2019-09-21:
Hi,
a marker for TDR could be useful to identify distance of junction ( or other anomaly).

73

mihtjel, 2019-09-21:
It sounds like an interesting idea! It probably needs the TDR graph to be scalable as well, if it's really to be of any value. I'll consider it :-)

gaionim, 2019-09-22:
I noticed that the full scale value is a function of the stimulus. Perhaps we could investigate further and set stimulus based on the desired length (10m, 100m etc)

gaionim, 2019-09-22:
stimulus full-scale
1-30m 1.3km
1-300m 133m
1-900m 44m
1-1500m  26m

frohro, 2020-05-31:
Is there a way to use the TDR to measure a 1000 foot cable that has a break somewhere in it, to determine where the break is?

zarath, 2020-06-25:
Marker is implemented.

Perhaps a documentation chapter does and don'ts for TDR are nice in future

closed 2020-06-25 by zarath