NanoVNA-Saver/nanovna-saver #21: Error when stimulus CW

gaionim, 2019-09-22:
**Describe the bug**
If fstart = fstop and set as reference -> division by zero error

NanoVNASaver 0.0.10a
Copyright (C) 2019 Rune B. Broberg
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
Settings: /home/mauro/.config/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "/home/mauro/git/nanovna-saver/NanoVNASaver/Chart.py", line 786, in paintEvent
    self.drawValues(qp)
  File "/home/mauro/git/nanovna-saver/NanoVNASaver/Chart.py", line 877, in drawValues
    x = self.leftMargin + 1 + round(self.chartWidth*(self.reference[i].freq - fstart)/fspan)
ZeroDivisionError: division by zero

**To Reproduce**
Steps to reproduce the behavior:
1. nanovna in CW fstart = fstop = 1513806 Hz
2. Click on connect -> start sweep
3. click on Set current as reference
4. See error

**Expected behavior**
No error.

**Screenshots**
[image]

gaionim, 2019-09-22:
PS
do not ask why nanovna is in CW ,  autosetup at startup.

mihtjel, 2019-09-22:
I've attempted to make sure the NanoVNA ends in a non-CW state when we start sweeping. It's not a great fix yet, so I'm leaving the issue open for a while longer.

zarath, 2020-07-04:
Stil valid?

gaionim, 2020-07-04:
Now CW is not valid:
Unable to parse frequency inputs - check start and stop fields.
For me is right and whe can close.

closed 2020-07-04 by zarath