NanoVNA-Saver/nanovna-saver #25: Not detected by USB/COM port

saintofinternet, 2019-09-29:
**Describe the bug**
The nanovna device is not being deteceted by the USB as a COM port device.

It is not responding at all on the PC 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

Psynosaur, 2019-09-29:
Hi there

Can you connect to the NanoVNA via a serial COM at all?

when connected via your favourite serail application type info.

```
info
Kernel:       4.0.0
Compiler:     GCC 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Architecture: ARMv6-M
Core Variant: Cortex-M0
Port Info:    Preemption through NMI
Platform:     STM32F072xB Entry Level Medium Density devices
Board:        NanoVNA
Build time:   Sep  9 2019 - 21:45:20
ch>
```

Please let us know

mihtjel, 2019-10-24:
No response.

closed 2019-10-24 by mihtjel