NanoVNA-Saver/nanovna-saver #28: Move hardware specific functions out of NanoVNASaver.py

nanovna, 2019-10-03:
Currently the NanoVNASaver class mixes UI code with hardware specific commands and functionality. This will be a problem for V2 which will have a completely reworked command interface; the 101 point limitation is also removed.

The architecture I suggest is to rename SweepWorker to VNAHW, move all details about the serial interface into VNAHW, and VNAHW will expose a (clean) API to the application that hides the details about the HW and simply expose sweep data. Kind of like the C++ API here: https://github.com/xaxaxa-dev/vna/blob/master/libxavna/README.md

SweepWorker.app field can then be removed, and the hardware and UI code can be completely independent from each other.

If no one offers to work on this I'll start work in a week or so. Please assign this issue if you do intend to work on it.

mihtjel, 2019-10-03:
I too have heard that there's a new project, unrelated to the original creators of the NanoVNA, making a device called "NanoVNA V2". There are many other VNAs as well :-)

I will be looking at the hardware interface and the structure of this, but as long as these new devices haven't materialised and aren't available for debugging, I'm not going to put a lot of work into it.

If you want to contribute to the project, I would prefer it to be from an account that can reasonably be linked to an identifiable person. :-)

nanovna, 2019-10-03:
We would like to get all the software ready before product release, so I'm going to start work on this in a separate repo and in the end you can choose whether or not to integrate the changes. V2 prototypes aren't available yet but the refractoring part is independent of the V2 hardware and will allow supporting any other VNA in the future. Overally the code structure is fairly tidy (unlike the NanoVNA firmware!) so it shouldn't be too hard.

gaionim, 2019-10-03:
as an example I read data from MR100 (clone of the SARK100) as well
in the main I put

from .utils import MR100Saver as NanoVNASaver

`
class FakeSerial(object):
    def __init__(self):
        self.is_open = True

class MR100Saver(NanoVNASaver):
    def __init__(self):
        super().__init__()
        self.instrument = Mr100(raw=True)
        self.serial = FakeSerial()
        #
        self.sweepStartInput.setText("3M")
        self.sweepEndInput.setText("30M")
        self.updateStepSize()

    def readFirmware(self):
        return self.instrument.readFirmware()

    def readValues(self, data):
        return self.instrument.readValues(data)

    def setSweep(self, start, stop, step=101):
        return self.instrument.setSweep(start, stop, step)
`

mihtjel, 2019-10-13:
Hardware abstraction layer was made.

closed 2019-10-13 by mihtjel