NanoVNA-Saver/nanovna-saver #29: Impedance measurement ??

UlrichD61, 2019-10-04:
**I'm facing some problems when mesuring impedances with NanoVNA.** 
I am using that smart little NanoVNA device primarily to measure impedances of coils, wound beads and ferrite core materials for EMC purpose. In general that seems to work pretty OK.

Now I found that wonderful nanovna-saver SW here and was quite happy to run it.! Thanks for that effort !

However there is a significant difference in the impedance values shown on the tiny Nano-LCD (CH0 Smith chart with impedance value in OHM) and on the Chart in NanoVNA-Saver (S11 +jX).
Or do I have a principle misunderstanding how to interprete those values ?

Aanother question, is anyone aware of a NanoVNA-firmware release that shows the Impedance as a trace? It is rather timeconsuming to scribble down the impedance on various frequency points.
 
Any help is very much appreciated !

mihtjel, 2019-10-04:
Hello Ulrich,
if you could provide some screenshots/pictures of the measurement both on the NanoVNA and in NanoVNA-Saver, it will be easier for me to compare what you mean.

For other questions, I think the nanovna-users group on groups.io is a much better place to ask :-)

UlrichD61, 2019-10-04:
Hello mihtjel, thanks for your fast response.

Attached 5 pictures. 
3 photos of nano-vna at different frequencies (20MHz, 100MHz and 250 MHz) showing 3 impedance values.
2 screenshoot from PC with markers at those frequencies.

From PC-screen I read:
~75 Ohms, ~540 Ohms and ~200 Ohms

From Nano.vna I read
149 Ohms, 253 Ohms, 544 Ohms 

Am I doing something odd here ?

[image]
[image]
[image]
[image]
[image]

Thanks !

mihtjel, 2019-10-04:
Hello Ulrich,
the 540 ohms looks about right. So there's that.

Looking at the location of your first marker at 20 MHz, it looks to me like it's very close to the 50 ohm circle of constant resistance. Going by that, I think I stand by the reading from NanoVNA-Saver that the resistance is 68 ohm. Quite why the NanoVNA shows it as 149 ohms is very confusing.  I haven't looked into the other two examples as closely - one is 544 vs 542 ohms, the other slightly more different at 200 vs 250.  But I think my software is right, and your NanoVNA is giving you wrong numbers, sadly.

What version of the firmware are you running?

UlrichD61, 2019-10-04:
[image]

Thanks for your thought aand checking the charts.

I'll try with some well-known components and doubl-echeck the results of both calculations.

mihtjel, 2019-10-04:
That firmware does seem *very* old. I don't know if the newer versions are necessarily recommended, but I know there have at least been many changes to it since then.

I look forward to hearing from you, should you find any errors :-)

UlrichD61, 2019-10-05:
Hi, me again,
after upgrading the Firmware to version "NanoVNA-H__900_ch_20191003" it seems to be a exact fit between LCD and PC values.
Thanks a lot !

gaionim, 2019-10-08:
glad to know, so you can close the issue.
I don't want to be rude, but it's best to keep open only the unsolved problems 
73

mihtjel, 2019-10-08:
Thanks, I will close it then :-)

closed 2019-10-08 by mihtjel