NanoVNA-Saver/nanovna-saver #33: logarithmic scaling of graphs

UlrichD61, 2019-10-07:
Hello, me again.

After quite successfully using NanoVNA for impedance measurement of EMC components I'm asking if it's possible to add an option for the graphs to display in logarithmic scaling. Ideally selectable for amplitude and/or frequency.
That logarithmic scaling is the industry standard when looking into datasheets and having same scaling in nanovna-saver would be very helpful.

just  a thought.....

Here an example of a ferrite-core in logarithmic scaling.
[image]

mihtjel, 2019-10-08:
I had log scaling of frequency on the list, but I had not yet considered log scaling of impedance as well. Interesting point. I will put it on the wishlist :-)

UlrichD61, 2019-10-08:
Thanks, you are a star !!

Ulrich 

⁣TypeApp for Android herunterladen ​

mihtjel, 2019-10-13:
Log scaling of the frequency axis is in for 0.1.3.

Log scaling of the data axis: Which charts would it be relevant for? I'm thinking only Q and R+jX make sense?

UlrichD61, 2019-10-13:
Hello
Yes I think so too. 

Most other graphs are in dezibel, that is already logarithmic scaling. 

Cheers & Thanks for all effort into that!

Ulrich 

⁣TypeApp for Android herunterladen ​

UlrichD61, 2019-11-08:
Hello mihtjel,
once more a lot of thanks for all effort on Saver 1.5.  I'm sure the whole community is extremely greatful.

Regarding the logarithmic scaling of frequency, is there any chance to devide the screen into "decades" rather than equidistance scaling with odd numbers ?
Having the classic way of showing decades e.g. lines at 1 MHz, 10 MHz, 100 MHz..... (or even fractions ) would make it more useful and compatible to the standard.

Just a thought :-)

mihtjel, 2019-11-08:
Hello Ulrich,
it's certainly on the TODO list, but I still need to figure out a good way to do it :-)

UlrichD61, 2019-11-30:
Hello, me again. 
Just two more enhencement ideas:  

It is great to have the S11 |Z| available in the chart-selection.
Would it be a problem to add S11 |Z| also into the **"Marker Settings"** to the have the actual impedance as number ?

Additionally a little "**Text-Window**" in the graph image would be ideal to allow commenting the image before saving.
 
Thanks !

PS:
...... the decade-devided logartithmic frequency scaling is still on my Christmas wish list :-).
See post from Nov. 8th.  Unfortunately I'm not a SW programmer to support any of the topics with some  coding.

UlrichD61, 2019-11-30:
sorrry, closed by accident and reopenened again

rjordans, 2021-02-27:
#377 should fix the logarithmic plotting party of this

closed 2022-12-12 by github-actions[bot]

ikatkov, 2023-03-22:
Arguably, this is different from what the OP has been asking for.
The point of having industry standard graphs is to be able to make quick visual comparison.

These two charts show the same data, but the look completely different.

[image]

[image]

When one compares their own local measurements with manufacturer's 
[image]

lack of proper log scale is very visible