NanoVNA-Saver/nanovna-saver #40: Connecting to NanoVNA with Sweep Paused Can App Crash

mydevpeeps, 2019-10-11:
**Describe the bug**
Connecting NanoVNA to NanoVNASaver with the sweep function paused on the NanoVNA results in an app not responding.

**To Reproduce**
Steps to reproduce the behavior:
1) perform a sweep on the NanoVNA only using the VNA itself.
2) on the NanoVNA, pause the sweep (stimulus -> pause sweep).
3) ensure the nanovna saver serial port control is in a disconnected state.
4) connect the NanoVNA to the computer using a USB cable.
5) click "Connect to NanoVNA" button under Serial port control.

**Expected behavior**
NanoVNA Saver makes a serial connection to the device, with the sweep function stopped.
(would be great if it read the current sweep data upon connect, but i'm not sure if that's technically possible)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Chrome/FF
 - Version 0.1.2

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
I can reproduce this quite often, but it doesn't happen every time. When it doesn't crash the app, it seems to just start a new sweep automatically, erasing the sweep that was on the display previously.

mihtjel, 2019-10-11:
Thanks, I'll have a look and see if I can reproduce that! What firmware version are you running on the NanoVNA? (Press "About" in NanoVNA-Saver after connecting successfully)

mydevpeeps, 2019-10-11:
It says "NanoVNA Firmware Version: NanoVNA version ?"

zarath, 2020-06-27:
Not reproducable with current firmware an NanoVNASaver version

closed 2020-06-27 by zarath