NanoVNA-Saver/nanovna-saver #45: Application failed to start

B4Q, 2019-10-17:
I received the message This application failed to start because no QT platform could be initialised  Could not load the QT platform plgin windows even though it was found.
/Users/leobally/Desktop/application failed to start.JPG.                                                             WIndows Vista Business service pack 2
QT 5.0.2 installed
computer dell optiplex 755

mihtjel, 2019-10-17:
How are you trying to run it? From a binary, or direct from the Python source?

B4Q, 2019-10-17:
Good afternoon 
I downloaded the nanovna -saver and clicked on to run as an administrator the naovna-saver v0.1.3 exe
se dump screen below

Leo Bally
Senior Advisor for Quality Assurance and Processes 
Company B4Q
www.b4q.eu
Email info@b4q.eu
Mobile 722932111
Stribrna Lhota 1021
Mnisek pod Brdy
Czech Republic

mihtjel, 2019-10-17:
Hello,
it doesn't seem like the attachment worked.  In general, Windows Vista is 2 years out of support from Microsoft, and I have no way to build binaries for it. If you want to run the software on Windows Vista, you might have to run it from the source Python code. See the instructions in the README for installing via "pip".

I don't know if PyQt5 is available in a suitable version for Windows Vista, but Python 3.7 should be.

Good luck! :-)

B4Q, 2019-10-17:
Thanks
I will give a try
Have a nice day

closed 2019-10-17 by mihtjel