NanoVNA-Saver/nanovna-saver #46: Smith chart

SMinin, 2019-10-17:
**Is your feature request related to a problem? Please describe.**

It would be very useful to add circles to the smith diagram for load matching. For example, like here https://i.stack.imgur.com/DPMa0.png. If there was a calculator for matching, it would be very cool. For example: https://www.will-kelsey.com/smith_chart/

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

mihtjel, 2019-10-17:
I think for now, there are much better programs for this, and some of them allow importing Touchstone files. There are other things in the application I will work on before I touch this :-)

SMinin, 2019-10-21:
It can only be supplemented with a mirror image of the Y chart, or make it possible to switch the Y chart and Z chart. This is very lacking when you matching the device with NanoVNA.

https://media.springernature.com/original/springer-static/image/chp%3A10.1007%2F978-981-13-3004-9_7/MediaObjects/468563_1_En_7_Fig31_HTML.png