NanoVNA-Saver/nanovna-saver #47: Precise movement between points

SMinin, 2019-10-17:
**Is your feature request related to a problem? Please describe.**

Lack of precise movement between points. Sometimes you need to look at information about neighboring points, and getting into them with the mouse is sometimes difficult.

**Describe the solution you'd like**
For example, using the keyboard cursor.

**Describe alternatives you've considered**

**Additional context**

mihtjel, 2019-10-17:
It's a good point. I think I might make up/down arrow in the marker frequency input fields move to the next/previous point. Would that fit?

PY3IG, 2019-10-19:
Additional Suggestion
A floating window next to the mouse indicating some information would aid selection.
Thanks!!

SMinin, 2019-10-21:
mihtjel, up-down or left-right arrows (it seems to me more intuitively) would be a great solution.

commit 183c1fbd (2019-10-30): Move markers step by step using up-arrow/down-arrow (Fixes #47)
closed 2019-11-05 by mihtjel