NanoVNA-Saver/nanovna-saver #48: Analytic information about the minimum of VSWR

SMinin, 2019-10-17:
**Is your feature request related to a problem? Please describe.**

Great idea with analytic information about the minimum of VSWR and S21 values. But sometimes there are more than one peak and it would be great to display information about all the peaks of S11 lows.

Sometimes you need additional information about this minimum and when you click on a line with a minimum, you want to move the cursor to the frequency of this minimum.

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Additional context**

mihtjel, 2019-10-17:
I like the idea of being able to move the cursor to the minimum in an easy way. Finding "other minima" requires a level of analysis I'm not currently competent enough to implement, but I'm sure it could be done, particularly if someone were to provide the algorithm ;-)

gaionim, 2019-10-19:
try with this

[ContribAnalysis.py.txt](https://github.com/mihtjel/nanovna-saver/files/3746731/ContribAnalysis.py.txt)

 Try to find  usable bands.
For now only set available  markers

SMinin, 2019-10-21:
mihtjel, the algorithm is very simple - move through a data set if the new value is less than the previous one, then we move down and look for a value that will be greater than the current one - this will be one of the minima. Then we move up the dataset in search of a new recession. In order to exclude rattling, we set the hysteresis. For example, if the hysteresis is 10 points, then when we find the next minimum, we compare it with the previous minimum and if they are at a distance of less than 10 points, then we save only the smallest minimum of these two.

gaionim, 2020-12-20:
VSWR Analysis show minimums,
can we close ?

closed 2020-12-21 by zarath