NanoVNA-Saver/nanovna-saver #49: PyQt5 QPalette & QColor not supported on raspberry pi - Always return Black Color

cinosh07, 2019-10-17:
When the application is started with the import of  QPalette & QColor from PyQt5 library, all colors are returned black.

But by using QPalette & QColor from qtpy library it worked.

Maybe an import conditional to the host type could solved the problem?

Regards

mihtjel, 2019-10-17:
This sounds very odd! I don't have a Raspberry Pi which has a display attached, but I think I can probably find one to debug it on. I'll have a look!

cinosh07, 2019-10-17:
Hi Rune,

you don't need one.
There's a virtual image for vmware or virtualbox on the rpi website.

https://www.raspberrypi.org/downloads/

mihtjel, 2019-10-17:
Interesting - I didn't know that was possible what with the different CPU type. Thanks, I'll give that a go. :-)

cinosh07, 2019-10-17:
You're welcome !  ;-)

Just tell me if you want me to check on my side your solution.

closed 2020-11-01 by zarath