NanoVNA-Saver/nanovna-saver #50: Export data to Excel

PY3IG, 2019-10-18:
I think exporting the data generated by VNA to Excel allows you to do other work.
As an example we have the study of G3TXQ with RF chokes.
With miniVNA you can reproduce this experiment using the spreadsheet available.
http://www.karinya.net/g3txq/chokes/
I would like to reproduce the experiment using nanoVNA
Thanks

mihtjel, 2019-10-18:
Hello,
the Touchstone files exported under "Files", "Save file (S1P)" / S2P can be imported into Excel. :-)

closed 2019-10-18 by mihtjel

PY3IG, 2019-10-19:
Could you help me name each column? This data was exported to S2P.
I want to capture the data loss transmission and transmission phase.
Thanks!
# | **Hz** | S | RI | R | 50
3000000 = **Hz** probably
0.9927071011657248 = 
0.06675267100972475 =
0.004743234093470405 =
-0.08028878868055044 =
0
0
0
0

mihtjel, 2019-10-19:
Hello,
the specification is here:
https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf

The format is:
HZ S11-Re S11-Im S21-Re S21-Im 0 0 0 0

The value represented is S11 and S21 in complex form. See:

https://en.wikipedia.org/wiki/Scattering_parameters

PY3IG, 2019-10-19:
Thanks! Spreadsheet already in operation!

mihtjel, 2019-10-19:
Cool! If you publish any results from it, I would love to see them :-)