NanoVNA-Saver/nanovna-saver #51: Error when installing with Loinux mint 19 32 bit

oe3jtb, 2019-10-19:
HI 
when I run
"python3.7 -m pip install ."

error
No matching ditribution found for PyQt5==5.11.2
I hav installed 5.10, but no chance to get 5.11
pse for help
tnx

mihtjel, 2019-10-19:
If you edit setup.py and remove "==5.11.2" so it just says PyQt5, it should be fine.

oe3jtb, 2019-10-19:
Hi tnx for fast answer
I did so, but the error message is the sam but without ==5.11.2
When I check pyqt version its 5.10.1+dfsg-lubuntu2 
Have I to qualify this name?
tnx

mihtjel, 2019-10-19:
Using the "pip install" methods install requirements from pip/PyPi. PyQt5 is not available there for 32 bit linux. If you remove it from setup.py, it should run anyway, assuming you have a compatible version installed.

oe3jtb, 2019-10-19:
so I should remove the whole statment from setup.py and run install again?

mihtjel, 2019-10-19:
You can remove the PyQt5 requirement from setup.py and run install again. If that does not work, install the requirements manually instead, and just run "python3 nanovna-saver.py".

oe3jtb, 2019-10-19:
yah ok, I did it
installation now is going through, but when running with python3.7 some errors according PyQt5 are appearing, when running python3 command "nomodul numpy etc,,," appears
So I believe, its not working with 32 Linux :-(

mihtjel, 2019-10-19:
If you don't run it using the "pip install" method, you need to make sure you install "numpy" and "pyserial" in addition to "PyQt5"

oe3jtb, 2019-10-19:
hmh I did 
apt install pyserial
apt install python-numpy   /* version 1.13.3 is displayed
thean I do the python3 command , same result...

mihtjel, 2019-10-19:
Okay, I don't know Linux Mint, whether it has a parallel python 2 install (which it might be pulling the packages for), etc.  But I do know that NanoVNA-Saver itself makes no assumptions as to whether it's running on Linux 32 or 64 bit - it merely requires Python 3.7, PyQt5, numpy and pyserial in order to run :-)

closed 2019-10-19 by mihtjel