NanoVNA-Saver/nanovna-saver #52: Window is shaking conditionally

PA3CCE, 2019-10-19:
Hello,
My name is Jos and my callsign is PA3CCE.
I just received my NanoVNA and it is flashed with the latest "NanoVNA-H" firmware.
On my Windows 10 system I'm running version 1.3 of your "Saver" software, that I find very promissing.
I found some strange behavior when I was measuring a 3.1 MHz to 5.2 MHz bandpassfilter :
When I selected the predefined 80 Mtrs settings all runs fine.
However,  the measuring results of the passband were shown only.
My filter is wider and I want to see the passband as well as the stopband. 
So I increased the span. Now it appeared that when I set the startfrequency at 3.25MHz it still runs fine, but when I set it at 3.2, 3.15 and 3.1 MHz the sweep starts ok, but when it reaches a certain frequency the whole screen starts to shiver heavyly, 
After some frequency the shivering stops and the sweeps continues normaly.
 Strangly, when I set the start frequency to 3.00 MHz all runs normally again.
I have a strong feeling that there is a problem in the calculations being excecuted by the program. When a sweep has finished and the results are still on screen I can activate the shivering by changing a marker frequenciy, so there is a relationship.
I also found the the parallel resistance of the input impedance of the filter increased strongly when the shivering occured, might also be related.
I hope the above may be of help.
Best regards
Jos
jrs@xs4all.nl

*Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mihtjel, 2019-10-19:
Hi Jos,
I *think* this happens because of marker data resizing the window, though I'm not sure why it would go back and forth unless you're clicking the charts in order to place the marker. It shouldn't usually do that.

I'm reworking that code for 0.1.4, so I hope it will go away there - otherwise, we will look at it again!

Thanks for reporting this :-)

mihtjel, 2019-10-24:
This didn't make it for 0.1.4, but it's on the list for the next release.

PA3CCE, 2019-10-24:
Same thing as with v 0.1.3 happened with v 0.1.4  , after using it several times finally the shaking has stopped.!
Incidently I have seen an erormessage that it was not possible to read frequencies !!
I prefer a darker color of green for the default Marker 3, this light one is hardly visible.

Thanks, Jos

cmoher3, 2020-03-07:
Neophyte user here. Just downloaded the latest version and clicked the .exe without connecting the nano. Thought I'd check out the gui. I got a gui display but all it does is violently twitch. I have a video of it doing that but I guess I can't upload here. I'm in Windows 10.

zarath, 2020-06-28:
Stil valid?

PA3CCE, 2020-06-30:
No, issue solved in Version 0.2.2.1
Jos

closed 2020-06-30 by zarath