NanoVNA-Saver/nanovna-saver #54: NanoVAN server on a MAC, white font over white background, hence text not visible.

ei8hvb, 2019-10-22:
Hello

I'm running NanoVAN server 0.1.3 on an iMac, and the GUI the inputs have a white font over a white background, and hence the text is invisible (I can see the text if I highlight it with the mouse). The software is otherwise running great, it's just this issue that the input texts, ie sweep control, markers, serial port, etc having a white font over a white background....

I can't see anything in the display setup that helps?

73

ei6ko

mihtjel, 2019-10-22:
Hello,
the application does not set the font colour, so it must be coming from somewhere within your display setup. Are you running a specific theme or similar on your iMac?

What version of OS X are you using?

I know there are other users (on the groups.io group) running NanoVNA-Saver on Mac, and who don't report these issues.

ei8hvb, 2019-10-22:
OK understood, thank you, I'm running macOS Mojave 10.14.6, I did try changing the system colours scheme.... but it didn't help, I will need to try & find out where the colours are set, perhaps its python setting.

thanks again for the reply, it will help narrow the search ...

Harry

mihtjel, 2019-10-22:
Hi Harry,
you might have some "Qt" or "PyQt" settings as well on your system. Those
could be relevant as well!

ei6ku, 2019-10-22:
I just tried the nanoVNA server on a Mac running High Serra 10.13.6, and its all as expected with black fonts over white background, so yes its specific to my macOS Mojave machine, or perhaps the PyQT on that machine.... but I suspect its the default dark colour scheme in macOS Mojave that at the route of this problem..... 

Thanks again for the input.....

Harry

ei8hvb, 2019-10-22:
actually I think it may need an explicit font colour to be set in "NanoVNASaver.py" 
in the macHD/user/xxxx/nanovna-saver/NanoVNASaver/NanoVNASaver.py 
as the default system settings in macOS Mojave are not appropriate ......

mihtjel, 2019-10-22:
Hi Harry,
What version of pyqt5 do you have installed? Have you tried disabling Dark
Mode in Mojave?

ei8hvb, 2019-10-22:
Thanks yes running NanoVNASaver in Mojave "Light" mode works, as a work around I'll just avoid the Dark mode for now..... the PyQt version on this machine is PyQt 5.9.2

Regards

Harry

mihtjel, 2019-10-22:
Hi Harry,
try upgrading to *at least* 5.12.0 - the fix for the light mode/dark mode
change is in 5.12.0 :-)

ei8hvb, 2019-10-22:
OK thanks - I'll get that updated, appreciate your time sorting this.

73

Harry

ei8hvb, 2019-10-23:
Thanks again, NanoNVAsSaver running under Python3 and PyQt 5.12.0 has fixed the problem.
For anyone else wishing to fix the same issue on a Mojave MAC this guide on installing Python3 is very helpful ...... 
https://docs.python-guide.org/starting/install3/osx/

73

Harry

mihtjel, 2019-10-23:
Hi Harry,
thank you very much for updating! It's great to hear that it's working! I
will try to put a note about this in the README, so I'm leaving the issue
open until I do!

mklengel, 2020-01-18:
The problem can be solved with a small correction in the file NanoVNASaver.py. Just commenting out some lines in the class AboutWindow, e.g.:
```
class AboutWindow(QtWidgets.QWidget):
    def __init__(self, app: NanoVNASaver):
        super().__init__()
        self.app = app

        self.setWindowTitle("About NanoVNASaver")
        self.setWindowIcon(self.app.icon)
        top_layout = QtWidgets.QHBoxLayout()
        self.setLayout(top_layout)
        # self.setAutoFillBackground(True)
        # pal = self.palette()
        # pal.setColor(QtGui.QPalette.Background, QtGui.QColor("white"))
        # self.setPalette(pal)
        shortcut = QtWidgets.QShortcut(QtCore.Qt.Key_Escape, self, self.hide)
```

zarath, 2020-06-28:
It's in v0.3.0

closed 2020-06-28 by zarath