NanoVNA-Saver/nanovna-saver #61: NanoVNAsaver keeps running after shutdown

PA3CCE, 2019-10-23:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

mihtjel, 2019-10-23:
What version did you run? When did you download it? What did you try to do? What failed? What are the symptoms?

If you downloaded 0.1.4 *immediately* when it was released, there was a bug in the first .exe uploaded, which only stayed up for a few minutes. Grab the download again, and it should be fixed.

PA3CCE, 2019-10-24:
Same problem in version 0.1. 3 and 0.1.4 (latest download) , when I shut down the program, makes no difference how, I can hear the program still running on my SDR (Software Defined Radio).

Update :
 It's not the program that keeps running, it's the NanoVNA, that's not switched of, sorry.