NanoVNA-Saver/nanovna-saver #62: Selecting several bands for sweep

pedjas, 2019-10-23:
For examining multiband antennas there is need to set sweep that covers larger frequency range than single band.

I tried creating my own "band" that covers several single bands (Band.ini) and it works, but then wide band is displayed as one band hiding single bands in the same range.

It would be good to have on option to select multiple bands (like using check boxes) and sweep range is calculated as lowest frequency of the lowest band and highest frequency of the highest band.

barjac, 2019-10-24:
I think that my suggestion in #59 would probably cover your use case.

pedjas, 2019-10-24:
Actualy there is difference. My suggestion is about setting sweep range to cover several preselected bands.

closed 2022-12-12 by github-actions[bot]