NanoVNA-Saver/nanovna-saver #67: Can't stop sweep at start of program

barjac, 2019-10-26:
**Describe the bug**
The last run failed with "Failed reading data 1 20 times."
The nanovna  disconnected when the dialog was closed.
Re-connecting the nanovna in order to adjust settings to resolve the issue instantly starts the sweep. There is no means of stopping this sweep and an endless loop begins.

**To Reproduce**
I am testing a 500MHz low pass filter intended for 70cms use.
If I set the band selector to "70cms" and run a sweep then all is fine.
If I then try to use the sweep settings dialog (start, stop etc.) I would expect the "band" selection to be ignored as I have just over-ridden it, and I see no other way of cancelling the band selection. Simply changing the sweep to cover 350 to 550 MHz and running the sweep causes this bug and there is no way back. The frequency marks on the graph axes do not change.
I have calibrated the nanovna manually using the cables and connectors used in the tests prior to this and run the same sweeps manually on the vna itself without issue.
See the attached mp4 video (zipped)
[nanovna-saver.zip](https://github.com/mihtjel/nanovna-saver/files/3774176/nanovna-saver.zip)

**Expected behaviour**
Re-connecting to the nanovna should not automatically start a sweep, allowing for setting changes to be made.

**Desktop (please complete the following information):**
 - OS: Mageia Linux 8(dev) running nanovna-saver-0.1.4
 - Browser Firefox
 - Version 68.2.0esr

**Additional context**
Please ask for any extra material - I hope the enclosed is adequate ;)

mihtjel, 2019-10-26:
Hi,
the error that NanoVNA-Saver failed reading data comes from the NanoVNA sending data that lies significantly outside the range that's expected. This has, in all the cases I have seen, because of the NanoVNA itself not being properly calibrated for the full span of frequencies, leading to invalid values. It is my experience that the calibration must, for consistent results, be saved as calibration setting number 0.

Note, that a full calibration of both NanoVNA ports is required.

NanoVNA-Saver reads the frequency range of the NanoVNA, and uses this to set up the first sweep range. Users expect to see data immediately after connecting their device, and the software therefore tries to fetch data immediately after connecting. Changing the frequency range on the NanoVNA can be uses to have the software read another range on first connecting.

barjac, 2019-10-26:
Thanks,
>> for consistent results, be saved as calibration setting number 0.
Right, I had saved the cal with the leads etc. attached to number 1. Recalling 1 and saving it to 0 before starting nv-s fixed the start-up issue.

>>  Changing the frequency range on the NanoVNA can be used to have the software read another >> range on first connecting.
Ah OK, now I know that the software pulls frequency data from the vna on connection, things may become less hit and miss. :)

As always, thanks for your help and a great piece of software.

Closing then.

closed 2019-10-26 by barjac