NanoVNA-Saver/nanovna-saver #68: Failed building wheel for NanoVNASaver

Twilight-Logic, 2019-10-26:
**Describe the bug**
I am getting the following error when running the install command:

 Running setup.py bdist_wheel for NanoVNASaver ... error
  Complete output from command /usr/bin/python3.7 -u -c "import setuptools, tokenize;__file__='/tmp/pip-bd9ju62f-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/tmp_eemos37pip-wheel- --python-tag cp37:
  /usr/lib/python3.7/distutils/dist.py:274: UserWarning: Unknown distribution option: 'long_description_content_type'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help

  error: invalid command 'bdist_wheel'

  ----------------------------------------
  Failed building wheel for NanoVNASaver
  Running setup.py clean for NanoVNASaver
Failed to build NanoVNASaver

**To Reproduce**
Steps to reproduce the behavior:
1. Download source code (nanovna-saver-0.1.4.tar.gz) package from https://github.com/mihtjel/nanovna-saver/releases/tag/v0.1.4, and unpack into a directory.
2. Check that dependencies present *see note
3. Navigate into the directory, e.g. cd nanovna-saver
4. python3.7 -m pip install .
5. python3.7 nanovna-saver.py

* the last dependency is described as * `numpy`. This is ambiguous. I had the 'python-numpy' package present, but when the install command finished and I tried to run with:

$ python3.7 nanovna-save.py

I got this:
python3.7: can't open file 'nanovna-save.py': [Errno 2] No such file or directory

Evidently the install did not complete correctly and the program file was missing, despite the concluding message telling me that the install had succeeded. I then installed 'python3-numpy', and although the 'bdist_wheel' error above persisted, the nanoVNA-saver program would now start and I could connect to the nanoVNA. However, I cannot be sure whether all of the functionality of the program is present.

**Expected behavior**
I expected the installer to complete without error and the program to then run using the command suggested in the readme.

**Desktop (please complete the following information):**
 - OS: Linux Mint 19.2 (Ubuntu 18.04LTS), with Cinnamon
 - Browser [Firefox, Chromium]
 - Version [FF69.0.1, Chromium  77.0.3865.90]

**Smartphone (please complete the following information):**
  n/a

mihtjel, 2019-10-26:
You ran the command "python3.7 nanovna-save.py", where you should have run "python3.7 nanovna-save**r**.py"

Twilight-Logic, 2019-10-26:
Thanks but that is weird, since I copied and pasted the command from the README.md. i didn't type it in, as I had in point 5 under 'To reproduce'. But yes, if I had ran it that way then it would certainly explain why it didn't run the first time! However, I think the typo is in the text, although I haven't got the session open to go back to and confirm.

However, that still doesn't explain the 'bdist_wheel' thing which happens regardless. What functionality will be missing in the program without this part of the process being completed? What do I need to do to clear this up?

mihtjel, 2019-10-26:
No functionality will be missing. I have confirmed that the spelling error is not in README.md, so I don't know where it came from.

I will try to download a VirtualBox image of Linux Mint and see if I can replicate the error you experienced once SourceForge downloads are back up, and if so, see if there is a way to avoid it.

Twilight-Logic, 2019-10-26:
mihtjel, thank you for looking into it.

I guess the spelling error was mine somehow and yes I agree it is NOT present in the README.md. Must have accidentally missed the trailing letter during copy or on the command line before hitting return.

I did a couple of tests here myself. I removed python3-numpy, leaving only python-numpy and ran nanovna-saver. That confirmed that it will run with python-numpy. I then did the opposite and removed python-numpy and installed python3-numpy. It also ran, which means that the program doesn't seem to mind which version is installed. Is there a preference? python3-numpy would of be consistent with the other two dependencies, but what would you advise?

The only thing left to note in connection with numpy is that 'sudo apt install numpy' does not work as there is no such package in the Ubuntu repository, so it needs to be specified as one of the above just mentioned. My apologies since I could have been more specific about that at the outset. 

The  'bdist_wheel' error, however, happens during the pip install and i appreciate your looking into it.

mihtjel, 2019-10-26:
I've checked using a virtual machine, and replicated the error you saw. It seems Mint (and possibly Ubuntu?) does not install python3-wheel by default, which causes a step of the installation to fail. The software still works fine though, and the dependencies are installed by the pip command. I will add a note to the README for the next release to install python3-wheel as needed.

mihtjel, 2019-10-26:
And thank you for reporting the problem, of course. :-)

closed 2019-10-26 by mihtjel

Twilight-Logic, 2019-10-27:
mihtjel, thank you. That was quick!

That does indeed solve it. In fact it was too obvious! I did look for something python wheel in apt search, but missed it, probably by focussing on 3.7 related packages. I have installed python3-wheel as indicated and ran the install. This time the install completed without error, so from my point of view that resolves the issue. The nanovna-saver application even appears to start faster.

Thank you for a very quick response!

Thank you also for making this application available to NanoVNA users. I have already found it helpful. It is so much easier to drive the NanoVNA from a bigger screen.