NanoVNA-Saver/nanovna-saver #70: Running a calibration causes abort.

PA3CCE, 2019-10-28:
**Describe the bug**
A clear and concise description of what the bug is

Hi Mihtjel,
I was running a calibration from 850k - 2M using version 0.1.4 of NanovnaSaver, When having finished " Short/Open/Load"  I selected "Apply",  This causes an abort of the software. I tried twice in a row.
Thereafter running version 0.1.3 and the same procedure : This did NOT happen.
Jos

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.1.4]

**Additional context**
Add any other context about the problem here.

mihtjel, 2019-10-28:
Hi,
did you use the calibration assistant? If not, please make sure you use that.

If you still have problems when using that, try starting NanoVNA-Saver from a command prompt (Assuming you are running Windows, as you did not say). This should preserve any error message from the crash, which you can then enter into this issue.

PA3CCE, 2019-10-28:
Hi,
When I start Nanovnasaver in Windows it is loaded from a DOS box .When nanovnasaver is running i select  that DOS box and at that time there are no error messages.
When I look into the DOS box after having performed the calibration, but not yet clicked the "APPLY" button, there still are no error message in the box.
When I have the DOS box overlayed and click the " APPLY"  button in nanovnasaver the DOS box and the program remain on screen for a second,  multiple lines are being written in the box.
I think the first lines are no errors just confirmations, but from the last line I could just read the end
 " ..... floating divide by zero" . Time is too short to read well, anyhow the problem is caused by this calculation error.
I hope this helps.
Jos

mihtjel, 2019-10-28:
Hi,
please start it by running it from cmd.exe - that way, the window doesn't
close, and the error messages remain available after the crash. :-)

Please also describe the situation more clearly: What steps did you take,
what buttons did you click in order to cause the crash?

PA3CCE, 2019-10-28:
Hi,
Did the following:

1) Started cmd.exe
2) Navigated to the directory that contained nanovnasaver 0.1.4
3) started nanovnasaver.v.0.1.4
4) Connected nanovnasaver to the device (nanovnasaver looaded the data from previous action).
5) Entered the start and stop frequencies (4MHz and 5MHz, but the same thing happens with others)
6) Clicked "calibration".
7) Clicked "Calibration Manager"
8) Followed the instructions "short, open Load"
9) After "Load" a Window pops up 
10) Clicked "Apply" in that window.
11) Nanovnasaver crashes !!
12) DOS box showing the following (I omit the lines that showed after starting nanovnasaver)

Traceback (most recent call last) :
   File "NanoVNASaver chart.py",  line 465, in paintEvent
   File "NanoVNASaver chart.py", line 2254, in drawValues
   File "NanoVNASaver chart.py", line 2373, in getReYPosition
ZerodivisionError:  float division by zero

After this a return to the prompt.
That's all,....success
Jos

mihtjel, 2019-10-28:
Excellent, thanks! I think I know where the error is in the code, and I can try to fix it for next time :-) You are ending up with a data set after calibration where the span of the resistive values is 0 - so when the chart (here I think the real/imaginary combination chart) tries to divide by the span, it ends up dividing by zero!

If you change the R+jX chart or Real/Imag chart out for something else it should not crash - you can then make your measurements, and re-enable the charts, and they should be fine.

I will fix the crash bug for the next version. :-)

PA3CCE, 2019-10-29:
OK, fine, yesterday I left it as it was and this morning I started again.
I loaded nanovnasaver, strangely it started with showing the smithchart only.
Then I pressed " Display setup " and saw all my selections were still present.
So I left "Display setup"  and then all S11 selected charts were showing again (smith chart, .R+jx , ReturnLoss, SWR, Phase and Quality factor).
After this I went back to "Display setup" and left just Smithchart, ReturnLos, and VSWR.
I did the calibration procedure as described in my privous message and after clicking "Apply ", nanovnasaver did NOT crache, however when I went back to the display screen all three charts showed "Data outside frequency span" 

I repeated the whole procedure, then the smith chart and the SWR showed the ideal values (50 Ohms and VSWR 1.0)  but the ReturnLoss chart  was filled with red lines from 0 - 40 dB's.
After that I saved the calibration.
When I exited nanovnasaver and started it again, just the smitch chart showed like above.

These are my observations sa far.
Jos

PA3CCE, 2019-10-29:
Aditional observations :

After sending the previous message Nanovnasaver 0.1.4 was still running showing the smith chart only, so I clicked " Disply setup" and back this resulted in showing the OLD setup with the 6 charts, I had to set the 3 charts again to " None".
After restarting nanovnasaver now it started up correctly with the three selected charts.

Jos

PA3CCE, 2019-10-29:
More info

I returned  to nanovnasaver , it was still running with 3 charts and Frequency 3 to 5MHz setting and did some measurements on my device (S11 on FuncubeDonglePro+), with results as expected.
Then I wanted to do the same measurements on a different frequencyspan so I set to 100k to 3M.

I wanted to callibrate this setting so executed the calibration procedure.
Pressing "Apply" caused crashing nanovnasaver again. Tried it twice.

Then I removed the R+ jX  display setting, tried again and successfuly calibrated.
Did a measurement on my device with rresults as expected.

Thinking all measurements are performed so let's try , I went to "Display settings" , tried to set "R+jX" again, this caused an immediate crash.

Might be of help,
Jos

PY3IG, 2019-10-29:
The same problem happened to me.
To "fool" the problem by following the calibration until the final (isolation and Through).
So clicking "apply" did not return the software crash.

mihtjel, 2019-11-05:
Excellent! I was meaning to ask you about this, but I'm happy that the fix has worked for you!

closed 2019-11-05 by mihtjel