NanoVNA-Saver/nanovna-saver #78: VSWRAnalysis show same value

gaionim, 2019-11-06:
VSWRAnalysis with more than 3 values shows 3 time the lowest

I expect you to show the lowest 3

[image]

If needed I can attach samples data

mihtjel, 2019-11-06:
This certainly looks like a bug!

For testing, I'd appreciate the data file, it would make things easier :-)

gaionim, 2019-11-06:
[sample.zip](https://github.com/mihtjel/nanovna-saver/files/3813521/sample.zip)

gaionim, 2019-11-06:
there is mismatch from index on dips and minimums  when you delete min value.
if you change the value so it is non more the minimum  seems to work

< dips.remove(np.min(dips))
> dips[np.argmin(dips)] = 1000

commit af37973c (2019-11-06): Bugfix: VSWR analysis should now find the correct dips when there are more than 3. Fixes #78
closed 2019-11-12 by mihtjel