NanoVNA-Saver/nanovna-saver #79: pep8 compliance

gaionim, 2019-11-06:
With some editor ( pydev for eclipse for sure ) You can force [pep8](https://www.python.org/dev/peps/pep-0008/) on save.

When I try to make little mods to your useful code, it ends with a lot of mods due to pep8
so it is difficult ( to me ) keep syncronization with upstream.

I can disable pep8, but the right way is to use pep8 ? ;-)
73

mihtjel, 2019-11-06:
I mostly stick to pep8, except for line length (120 characters instead of 80, we're not on VT100 terminals any more :D)

There are a few places where variable or class naming is intentionally in breach of pep8. And a number of places that need to be cleaned up, but haven't been yet. As I re-write code or implement new features, I try to avoid the warnings that PyCharm give for pep8 breaches.

I will write code guidelines for the project at some point :-)

gaionim, 2019-11-06:
> I mostly stick to pep8, except for line length (120 characters instead of 80, we're not on VT100 terminals any more :D)

there are others "little" as spaces after # or so  ...

> There are a few places where variable or class naming is intentionally in breach of pep8. 

is not case of autopep8, it do not change names

> And a number of places that need to be cleaned up, but haven't been yet. 

I think autopep can be configured to ignore 80 char and  let it do some work ?!?

> I will write code guidelines for the project at some point :-)
ok
73

zarath, 2019-11-12:
I use pylama / pylint...
and yes one can set it to ignore 80 character boundary...

But I like it if I can manage to keep that boundary :-)

closed 2022-01-04 by zarath