NanoVNA-Saver/nanovna-saver #80: nanoVNA measurement random spikes using segments (more than 1)

MaxCrt, 2019-11-06:
Hi,
since I have a Lab with several instruments, I performed the calibration with a professional Rosenberger toolkit. But the spikes were still visible.
Those spikes are positive (up to 45 dB over the noise floor; i.e -70 dBm of noise floor, -30 dBm of spikes). I tried changing the frequency range, the number of segments (multiband sweeps) from 1 to 15.
After further checks I noticed that the spikes frequencies are quite random (though in the RF input of nanoVNA there is a 50 Ohm calibrated load) and they change sweep by sweep.
Testing a frequency range of 10 MHz with multiband sweep (15 segments) there are more spikes than using a lower number of segments. With one segment only there is no spike.
So it seems the spikes are a consequence of multi segments sweeps.

The FW is 0.31
Could you verify?

Thank you.

BR,
Max IK1RGI

mihtjel, 2019-11-06:
Hello Max,
I am aware of at least one other user experiencing these problems with the 0.3.1 firmware. I don't know exactly why, but as the application is merely displaying the data it receives, and I have seen no indications that it does this incorrectly, I believe the error lies in the firmware.

I would suggest maybe trying the NanoVNA-Q firmware?

MaxCrt, 2019-11-06:
It could be the error lies in the firmware.
But if the error was in the FW, I don't think the number of spikes would increase more and more depending on number of segments (nanovna-saver) required in the measurement (if there is no spike in a single segment measurement).
With one segment only (using nanovna-saver, trying different frequency ranges) I never see any spike, with more than one segment the number of spikes increases

mihtjel, 2019-11-06:
NanoVNA-Saver repeatedly asks for 101 point segments, and the spikes that I have seen other users have problems with have been at the very first point in new segments - stemming from the fact that the firmware sends the wrong data for the first point in the segment.

I don't see any problems with the 0.3.0m firmware I am running, even when doing multiple segments. So it does not seem to be a problem in NanoVNA-Saver, though I'd welcome anyone wanting to investigate further.

MaxCrt, 2019-11-06:
I will try downgrading (0.3.0) or upgrading (now 0.4.0) the fw of nanoVNA and let you know.

MaxCrt, 2019-11-08:
I upgraded the nanoVNA FW to 0.4.0 and performed tests using nanoVNA-saver 0.1.5 (single and multisegment freq-range scan). Now spikes that were visible using FW 0.3.1 aren't anymore.
However stability has to be improved as well.

mihtjel, 2019-11-08:
Good to hear that it was just a firmware error.

What do you mean "stability has to be improved as well"?

closed 2019-11-11 by mihtjel