NanoVNA-Saver/nanovna-saver #82: Strange behaviour marker settings

PA3CCE, 2019-11-06:
Hi Mihtjel,

Thanks for the nice update, the calbration is not cashing anymore. but I still would be happy when
the desciption of the active calibration was shown in the maun window,

I'm very sorry to have a new issue:

**Describe the bug**
**To Reproduce**

1) Close NanoVNAServer after a scan with Marker1 set whith a frequency.
2) Restart NanoNAServer and click "Connect to NanoVNA"
   Now data from the  previous scan are going to be displayed, EXCEPT the marker - frequency
   and data.
3) Enter ANY marker frequency , OTHERS, then the one in the session before the restart
    of NanoVNASaver and IN THE SAME THE FREQUENCY SPAN being displayed and using 
   THE SAME CHARACTER  "H", "K", or "G" . (you might have to reread this sentence HI).
   After entering the character, the OLD marker- frequency settings and data OF MARKER 1 is
   and maintains being displayed, even after a new scan IN ALL MARKERS when frequencies
   as described are entered in them.

I hope I described it a bit anderstandable HI.
Jos

**Expected behavior**
A clear and concise description of what you expected to happen.

Looks clear to me.

PA3CCE, 2019-11-06:
I restarted nanovnasaver again and found that it now it started op with 50k to 900M range.
I'm pretty sure it started up earlier in the old range.
when resetting to a smaller range it seemed to be OK, But will investigate more.
Jos

PA3CCE, 2019-11-06:
Hi Mihjtel,

I have several several 50 Ohms standards and wanted to test them at low and high frequencies in an attempt to find out which one is the best.

So I started VNASaver and started a low frequency measurement,
I had set to 10 segments, the start frequency to 50 kHz and the stop frequency to 100 kHz,  this was the cause of the problem.
When I enter  51 kHz or higher in Marker1 at this span, VNAServer starts and keeps showing 50 kHz in the data field and the entered frequency in the marker box.  That made me think it was the old data.

When I set the stop frequency at 135 kHz the data field starts showing 59,9 kHz and Marker1 still keeps showing 60 kHz in its box.

When I set the stop frequency to 150 kHz all looks good, both (rounded)  frequencies are the same.

So, to my opinion, it's a combination of the low frequency and the small span  Though, VNASaver with stopfrequency set at 100kHz, correctly shows 50Hz/step, that misleaded me.

Another thing, can you tell me what is the cause of these long downward lines at these low frequencies , is this caused by missing data or noise on the signal ?
This is wondering me for long, I tend to think it's missing data, for at high frequencies it looks like the downward noise is  more than the upward also. I think this can be important when averaging.

I have another remark : 
The green progression bar stops after the first sweep, when "Continuous sweep" is activated.

Sorry for my dumb remarks and questions,  thanks in advance for your possible answers.
Jos

mihtjel, 2019-11-06:
Hello Jos,
what firmware version are you using?

Some (older) firmwares don't allow short step sizes, below maybe 1kHz. If
consecutive points have the same frequency, the software gets somewhat
confused.

PA3CCE, 2019-11-06:
Hello Mihjtel,

I downloaded the firmware recently 1/11/2019 and its called Nanovna with scan and 1500MHz.dfu dated 14/9/2019. I found it via the nanovna users group and to my best knowledge its from eddy555.

Jos

mihtjel, 2019-11-07:
15 September is way old by now 😉 there is a recent 0.4.0 release that might be worth trying.

PA3CCE, 2019-11-07:
Is it Really ?, I'll try to find it and let you know the resuls.
Jos

PA3CCE, 2019-11-07:
Well, I found one named 10kHz-1.5GHz_ 05.10..2019.dfu  is this the latest one, sofar I don't see a version 4.0.
Anyhow I will flash this one and see what happens.
Jos

mihtjel, 2019-11-07:
https://github.com/ttrftech/NanoVNA/releases

I recommend taking this to the nanovna-users groups.io group, though -
github issues aren't really a support forum. :-)

PA3CCE, 2019-11-07:
Yes in the mean time I  searched some more and found it, the name 0.4 misleaded me,  It's not a .dfu file the ,hex will load in dfu mode I guess,  I'll let you know the results.
Jos

PA3CCE, 2019-11-08:
Hi
I did not dare to use the .hex file in Dfu  SeDemo and did not succeed in compiling the hex2dfu software.
However, I found a file called 10kHz-1.5GH.05.10,2019.dfu and flashed this one.

Now, without loading a calibration file in NanoVNASaver,  I measured  a returnloss of > -50dB  over a frequency range of 10 to 100 kHz on the 50 Ohms standard and after calibration with this standard,
I measure a returnloss of  > 70dB on it.
I see no downward lines in the returnloss window ! Fantastic.

When I set a frequency in marker 1, this freq. setting remains in the box after a sweep but the data shows the nearby calculated one,
When clicking in the Returnloss window, both show a nearby frequency but not neccessaraly the same., no problem.
What are thes selection rounds used for, the marker setting works well without selecting them.

Thanks for this wonderful software !!!
Jos

PA3CCE, 2019-11-08:
Hi
With the new firmware now I have set the frequency range from 800M to 900M and calibrated with the 50 Ohms standard.
When I measure the standard with this calibration, I  measure 40dBs of returnloss with noise upwards let's say 35dB and downwards upto lets say 70db.
I am still wondering why the strong downwards noise.
In the R+jx chart all noise is equaly around r=50 Ohms and x=0

PA3CCE, 2019-11-08:
Dear reader,

It looks like the downwards lines in the returnloss chart  I see now at the higher frequencies has the same cause as with my previous loaded firmware at the lowest frequencies.
Nanovna firmware version 4.0  (ch.hex etc.) is dated 06/11/2019 and as said the firmware I have loaded now  is from 05/10/2019 so who kmows ?

Could anyone send me the 4.0 version in DFU format or give me an address where to find this file, or hex2dfu.exe ??
My email address is:   jrs@xs4all.nl

Many thanks in anticipation
Jos

mihtjel, 2019-11-08:
Hi Jos,
please take a look at the NanoVNA-Users groups.io group.  These bug/issue
reports are not a support forum, and I am satisfied for now that it is not
an issue with NanoVNA-Saver, but rather either with your device, or the
firmware you are running :-)

PA3CCE, 2019-11-08:
OK,, I, I understand and Iwill put the question in the users group.
In the mean time I found Nanovna-Q-0.4.3 in DFU format there and flashed it, same resullts.
It is related to the NanoVNA-H firmware, the release number made me think it was 4.0 related.
I don't think it's my hardware, its a black one with the metal screens.
That 4.0 release  you mentioned and where NanoVNASaver is based on, is  that the one by edy555 ??
The many variants makes me dizzy.
Jos

PA3CCE, 2019-11-09:
Hi Mihtjel,

I looked at Edy555's github page and saw his latest version was 0.4.0 so I think you were referring to this firmware indeed.

I also have been looking into the nanovna users group, most people are talking about version 2, none of them about version 0.4.0
When I look in the "files"  aeria there I see no versions 0.4.0 either, the highist version from EDY555 is " version 2.3 with modified scan command and negative edelay.dfu.
I downloaded and flashed this one

When I swith on the device or click  "Recall/Version"  it shows among others  "Version 2.3  Build Time: Oct 11 2019 - 09: 15:56"  and  "Kernel 4.0.0", seeing this build time I think I have the firmware you were referring to,
I still find it strange that on Edy555 the his latest version  is named as version 0.4.0

When I use it in NanovnaSaver the results at the lower frequencies in earlier flashed firmware were better if I remember well. But they are acceptable, the average retunloss is below 50dB's when I measaure the 50 Ohm standard.
The Touchscreen functions surely are worse, but that is no problem in NanoVNA-Saver.

I found that the rounds near the marker settings are used for setting the marker with the mouse.
The marker setting and data showed at the marker frequencies have a strong relation  with these downward lines in the returnloss chart.

My conclusion is that my problems indeed are firmware related, I well put them at the right place.
Thanks a lot for your time,
Jos

mihtjel, 2019-11-09:
Hi Jos,
thank you for taking such an interest in the software, and for working hard to get it to work - sounds good that you have a working solution now, and I hope you will continue to endeavour to improve your measurements :-)

closed 2019-11-09 by mihtjel