NanoVNA-Saver/nanovna-saver #85: Calibration issue

RonaldNC, 2019-11-08:
I'm using the new version and am having trouble with calibration.  Once I go through all the standards and try to apply them, I get an error that 2 of my open, closed or short values are the same (doesn't specify which 2).  I don't get this error when calibrating the unit directly.

Anyone else having this issue?

Ron

mihtjel, 2019-11-08:
Hi,
you need to either use the calibration assistant (Recommended), or sweep manually in between pressing the short/open/load/etc buttons - they don't run a sweep for you!

RonaldNC, 2019-11-08:
Thanks!  Where can I find the calibration assistant?  I've just been pushing the "calibration" button at the bottom of the home screen and completing the calibration.

Ron

mihtjel, 2019-11-08:
It's right there on the calibration screen - under the short/open/.. buttons

closed 2019-11-11 by mihtjel