NanoVNA-Saver/nanovna-saver #87: Window not opening

RonaldNC, 2019-11-08:
Sorry to be a problem... but I've been using a couple of days with no issues... but I rebooted my PC and now the script runs in a DOS window, but doesn't open the home window, then crashes.

Any ideas?

Thanks,

Ron

mihtjel, 2019-11-08:
Hi,
try starting the application from a command prompt (cmd.exe) and looking at what error messages it gives when it crashes.

RonaldNC, 2019-11-08:
Here's what I got...

C:\>nanovna-saver.v0.1.5.exe
NanoVNASaver 0.1.5
Copyright (C) 2019 Rune B. Broberg
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
Settings: C:/Users/Ronal/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "nanovna-saver.py", line 20, in <module>
  File "NanoVNASaver\__main__.py", line 73, in main
  File "NanoVNASaver\NanoVNASaver.py", line 355, in __init__
  File "NanoVNASaver\NanoVNASaver.py", line 2250, in __init__
  File "NanoVNASaver\Analysis.py", line 71, in __init__
IndexError: list index out of range
[876] Failed to execute script nanovna-saver

mihtjel, 2019-11-09:
I think you might have a setting set that the application is not handling
nicely. Possibly if you removed all the markers?

Try renaming or moving the .ini file it mentions on startup, and seeing if
that makes it start. I will take a closer look at the code tomorrow!

RonaldNC, 2019-11-09:
Figured it out.  There was a corrupt file in a hidden Windows app data directory.  Deleted them all and it works now.

Thank you for your help!

Ron

closed 2019-11-09 by mihtjel