NanoVNA-Saver/nanovna-saver #88: Please add buttons for recall calibration 0 to 4 into the device and set the sweep

yo2naa, 2019-11-09:
**Is your feature request related to a problem? Please describe.**
I have my NanoVNA calibrated on different frequency intervals, using all five calibration memories.
Sometimes, when using nanovna-saver I need to recall a different calibration memory into the device. It would be nice to be able to do this from the nanovna-saver.

**Describe the solution you'd like**
Please add five buttons to recall calibration 0 to 4 into the device, and also set the sweep start and stop accordingly

**Describe alternatives you've considered**
To recall calibration from the device (I need to change glasses) and to set the sweep manually on the nanovna-saver.

**Additional context**
N/A

zarath, 2019-11-10:
A drop-down instead of 5 fixed buttons could be considered to load stored messurement setups in future.

khaytsus, 2020-04-03:
The feature in NanoVNA WebApp (the Android app) to load from the device is super useful, as I have personally set up views, ranges, and calibrations for 3 different frequency ranges and a few other setups and I can quickly load them right on the phone as I analyze.  I have yet to figure out how to do this in NanoVNA Saver.  I load it on the device itself, then change the sweep settings.  That works, but it'd be nice to have a button right on the main screen to load settings from the device.

closed 2022-12-12 by github-actions[bot]