NanoVNA-Saver/nanovna-saver #93: Program crash after clicking on a chart

n7gtb, 2019-11-16:
**Describe the bug**
Program crash following clicking on a chart. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run a sweep. In my case I swept a 9 MHz bandpass crystal filter...
2. Once finished, click on any one of the charts.  It also happens when the chart has been popped out.
3. Program terminates.

**Expected behavior**
Marker should appear on the chart graph at or near the point clicked.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.1.4]

**Additional context**
Here is the full text provided in the console after crash (division by zero error):

This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
Settings: C:/Users/Vern/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
c:\users\vern\appdata\local\programs\python\python38-32\lib\site-packages\NanoVNASaver\Chart.py:3572: RuntimeWarning: divide by zero encountered in double_scalars
  delay = (-phase_change / (freq_change * 360)) * 10e8
2019-11-15 17:42:12,511 - NanoVNASaver.SweepWorker - ERROR - An exception occurred reading data 1: could not convert string to float: '-..001526420'
Traceback (most recent call last):
  File "c:\users\vern\appdata\local\programs\python\python38-32\lib\site-packages\NanoVNASaver\SweepWorker.py", line 328, in readData
    if float(a) < -9.5 or float(a) > 9.5:
ValueError: could not convert string to float: '-..001526420'
2019-11-15 17:42:12,512 - NanoVNASaver.SweepWorker - WARNING - Got a non-float data value: -06600504966 0.016690906 (-06600504966)
2019-11-15 17:42:18,011 - NanoVNASaver.SweepWorker - WARNING - Got a non-float data value: -06000241065 0.000005756 (-06000241065)
2019-11-15 17:42:20,858 - NanoVNASaver.SweepWorker - WARNING - Got a non-float data value: -0.000443021 -09300291294 (-09300291294)
Traceback (most recent call last):
  File "c:\users\vern\appdata\local\programs\python\python38-32\lib\site-packages\NanoVNASaver\NanoVNASaver.py", line 722, in markerUpdated
    m.updateLabels(self.data, self.data21)
  File "c:\users\vern\appdata\local\programs\python\python38-32\lib\site-packages\NanoVNASaver\Marker.py", line 398, in updateLabels
    self.s11_group_delay_label.setText(str(SITools.Value(RFTools.groupDelay(s11data, self.location), "s", fmt)))
  File "c:\users\vern\appdata\local\programs\python\python38-32\lib\site-packages\NanoVNASaver\RFTools.py", line 170, in groupDelay
    val = -delta_angle / math.tau / (freq1 - freq0)
ZeroDivisionError: float division by zero

mihtjel, 2019-11-16:
Hi,
I think this error comes up because you are running a step size that ends up less than the minimum distance supported by your firmware. Some older firmwares, at least, only support down to 100Hz - meaning some data points would be overlapping.

If you upgrade to, for example the NanoVNA-Q firmware, you should not see this error. It support 1Hz step size.

I will add putting in error handling for this error in the code backlog, as it obviously shouldn't cause a crash. :-)

n7gtb, 2019-11-20:
Thank you for addressing this issue.  Nanovnasaver is a great piece of work! :-) 👍 

Update: I loaded the NanoVNA-Q firmware (i.e., NanoVNA-Q-0.4.3-20f33ba.dfu), and the problem still exists.  Step sizes/frequency span combinations that result in less than 100Hz still cause nanovnasaver to end abruptly...

mihtjel, 2019-11-20:
Could you try the following: Start up NanoVNA-Saver without any markers, sweep a trace that you "know" would make it crash, and then save an S2P file of that data and upload it here?

That would let me peek into what exact data the program is seeing. :-)

n7gtb, 2019-11-20:
'crash-test.s2p' is attached.  As mentioned before, any frequency range and step combination that results in less than 100Hz/Step appears to cause the problem.Here is a quick excerpt from my notes: ****************************************Start: 8.98500MHzStop: 9.00992MHz2 seg, 123.4Hz/step No error
3 seg, 82.24Hz/step crash.

reloaded nanovnasaver, ran the sweep, and noticed this in the command window:
NanoVNASaver 0.2.0
Copyright (C) 2019 Rune B. Broberg
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
Settings: C:/Users/Vern/AppData/Roaming/NanoVNASaver/NanoVNASaver.ini
C:\Users\Vern\AppData\Local\Temp\_MEI124002\NanoVNASaver\Chart.py:3572: RuntimeWarning: divide by zero encountered in double_scalars****************************************
As a further test, I re-ran the sweep using the same start/stop frequencies and step=3 above with nothing attached to the NanoVNA, then attempted to set a marker.  This also reproduced the problem.
Hope this helps!
Best regards,Vern

mihtjel, 2019-11-20:
Thanks, this is very helpful!

mihtjel, 2019-12-04:
This should be fixed in the latest release.

closed 2019-12-04 by mihtjel