NanoVNA-Saver/nanovna-saver #99: Casual error

d51r3verse, 2019-11-18:
Really happy with nanovna-saver. Thnx for your knowledge chantry.
Followings is casual issues for me.

1. There are [no reference](https://github.com/mihtjel/nanovna-saver/blob/master/NanoVNASaver/Hardware.py#L203) in use as for now. But it should be to `data 0` 

2. Eliminating "Double calibration"( Both Device and PC side )
Even device in `cal reset` or `call off`, calibration status always "Device calibraion"( Does not seems to be using `data 2...6` )
If PC calibration in use, I think `cal reset` or `cal off` to the device before `sweep` or `scan` then pull `data 0...1`

3. Better touchstone file compatibility
scirf-kit generated touchstone format does not accepted( Ending with `50.0 `(trailing last space)
[point1](https://github.com/mihtjel/nanovna-saver/blob/master/NanoVNASaver/Touchstone.py#L56), [point2](https://github.com/mihtjel/nanovna-saver/blob/master/NanoVNASaver/Touchstone.py#L75)
For better [compatibility](https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf#page=6), `"^\s*#\s*(.?HZ)(?:\s+S)?(:?\s+(RI|MA))(:?\s+R\s+([^\s]+))?\s*$"`

5. Loading touchstone files, It seems read with writable mode I think
nanovna-saver always touching change time even just load s2p file.

mihtjel, 2019-11-18:
Thanks, this is a great report! I'll make sure these get fixed for the next release!

Re: 2) I want the user to have the option of calibrating on top of an existing NanoVNA calibration - but adding the option to calibrate on the bare hardware is definitely in the cards! You're right that the software should check the cal status if possible, though.

zarath, 2019-11-18:
Regarding Point 4, can you supply an example touchstone file?

mihtjel, 2019-11-18:
I'm sure you mean point 3, @zarath?

Re: Point 4 - the line opening the file opens in mode "r", which I took to mean "read only". My files certainly don't change their modify time stamp when I load them on Windows.

zarath, 2019-11-18:
You're right. It's point 3

d51r3verse, 2019-11-18:
> You're right. It's point 3

Simply use built-in [nanovna.py](https://github.com/ttrftech/NanoVNA/blob/master/python/NanoVNA-example.ipynb) script.

zarath, 2019-11-18:
I have rewritten the touchstone parser as in feature https://github.com/mihtjel/nanovna-saver/tree/feature/touchstone_rewrite.
But it has to be tested and polished before I would do a pull request

zarath, 2019-11-19:
I've left a comment in NanoVNASaver.py where a call to safe happens after touchstone load

zarath, 2019-11-20:
Ooops ... that scikit file is somewhat out of standards spec ....
at least regarding spec 1.1 which it is because the version keyword is missing.

Spec 1.1 requires int value for resistance in header, disallows comments after the header and requires monotonic ascending frequency values ....

Neithertheless I'll add possibilities in the parser to deal with that issues .... in near future...

d51r3verse, 2019-11-22:
> Ooops ... that scikit file is somewhat out of standards spec ....
> at least regarding spec 1.1 which it is because the version keyword is missing.
> 
> Spec 1.1 requires int value for resistance in header, disallows comments after the header and requires monotonic ascending frequency values ....
> 
> Neithertheless I'll add possibilities in the parser to deal with that issues .... in near future...
> 
> P.S . do you like to file a bug report to the scikit authors? :-)

Did not find any limitations about 'Reference resistantce must be integer' 'comments line posion'
Specifications pages says

` n is replaced by a positive integer or floating-point number` - like skrf generated file.

`keyword by at least
one whitespace character` `Option line parameters are separated by one or more whitespace` -like TAPR generated file.

`Comments may appear on a separate line, or after the last data value on a line`

d51r3verse, 2019-11-22:
@mihtjel 
Hidden `cal data` command will get device real calibrations terms.
https://github.com/ttrftech/NanoVNA/blob/master/main.c#L1376-L1381
Nanovna-saver will know device calibrated or not.

zarath, 2019-11-22:
> Did not find any limitations about 'Reference resistantce must be integer' 'comments line posion'
> Specifications pages says ( sorry for my bad english )
> ` n is replaced by a positive integer or floating-point number` - like skrf generated file.

I was refering to
https://ibis.org/connector/touchstone_spec11.pdf
and
https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf
> ` n is replaced by a positive integer or floating-point number` - like skrf generated file.
> 
> `keyword by at least one whitespace character` `Option line parameters are separated by one or more whitespace` -like fixed columned ( TAPR or [Keysight](http://literature.cdn.keysight.com/litweb/pdf/genesys200801/sim/linear_sim/sparams/touchstone_file_format.htm) generated ) file.
> 
> `Comments may appear on a separate line, or after the last data value on a line`

> > Ooops ... that scikit file is somewhat out of standards spec ....
> > at least regarding spec 1.1 which it is because the version keyword is missing.
> > Spec 1.1 requires int value for resistance in header, disallows comments after the header and requires monotonic ascending frequency values ....
> > Neithertheless I'll add possibilities in the parser to deal with that issues .... in near future...
> > P.S . do you like to file a bug report to the scikit authors? :-)
> 
> Did not find any limitations about 'Reference resistantce must be integer' 'comments line posion'
> Specifications pages says ( sorry for my bad english )
> 
> ` n is replaced by a positive integer or floating-point number` - like skrf generated file.
> 
> `keyword by at least one whitespace character` `Option line parameters are separated by one or more whitespace` -like fixed columned ( TAPR or [Keysight](http://literature.cdn.keysight.com/litweb/pdf/genesys200801/sim/linear_sim/sparams/touchstone_file_format.htm) generated ) file.
> 
> `Comments may appear on a separate line, or after the last data value on a line`

I was refering to
https://ibis.org/connector/touchstone_spec11.pdf
and
https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf

That keysight file is at least 2.0 because V1.1 didn't know about more then 4 ports etc...

d51r3verse, 2019-11-22:
> I was refering to
> https://ibis.org/connector/touchstone_spec11.pdf
> and
> https://ibis.org/touchstone_ver2.0/touchstone_ver2_0.pdf
>
> That keysight file is at least 2.0 because V1.1 didn't know about [...]

'Specification version' does not equal 'File version'.
'File version 1.0' corresponds 'Specification 2.0' or deprecated 'Specification 1.1'
'File version 2.0' corresponds 'Specification 2.0' only

mihtjel, 2019-11-22:
File handling should parse liberally, and save conservatively. 😊 So let's handle the types of files we see in the wild, and save files that are within the specs.

zarath, 2019-11-22:
> 'Specification version' does not equal 'File version'.
> 'File version 1.0' corresponds 'Specification 2.0' or deprecated 'Specification 1.1'
> 'File version 2.0' corresponds 'Specification 2.0' only
> TAPR, NanoVNA(SW), nanovna-saver, ... output(File version v1.0) also corresponds 'Specification 2.0'

What do you mean with file version?

Ibis TS v2.0 doc tells:

> [Version]
Rules for Version 1.0 files:
The [Version] keyword is not permitted for Version 1.0 files.
Rules for Version 2.0 files:
The [Version] keyword is required for files using any Version 2.0 syntax.

zarath, 2019-11-22:
> File handling should parse liberally, and save conservatively. blush So let's handle the types of files we see in the wild, and save files that are within the specs.

Agree, but TS2.0 is so far differnt from TS1.1 that It doesn't make sense to mix header parsing with my current class. I'ld would prefer to subclass the parser and implement the additions there....

also some methods are already prepared to cope with virtually unlimited number of ports...
eg. .s method with field order

mihtjel, 2019-12-04:
I think we are done with these issues with the latest release? Otherwise, let's branch any individual remaining issues to new issues instead of continuing a long thread. :-)

closed 2019-12-04 by mihtjel