NanoVNA-Saver/nanovna-saver #104: Comparison of >=2 measurements

DE8MSH, 2019-11-21:
Hey.

This is no issue. It's an enhancement.

It would be cool to compare >=2 measurements. Do m1, save it. Change something. Do m2, save it. Then load m1 and overlay the datapoints with m2. In another window maybe?

Just an idea.

mihtjel, 2019-11-21:
Hi,
are you aware of the "reference" feature? "Set current as reference" in the main window, or the "load as reference" feature under files.

DE8MSH, 2019-11-21:
Hey mithjel.

Oopsy. No. Will give it a try as soon as I'm at home again.

DE8MSH, 2019-11-22:
Hey mithjel.

I think I'm okay with the mentioned functions. I can now verify two measurements. I had in my mind that I need more than two. But I think two is okay.

Thank you for giving me the idea.

bvernoux, 2020-01-26:
I confirm It will be very nice to support more than 2 measurements especially to compare different measurements (on different design version) quickly