NanoVNA-Saver/nanovna-saver #114: Python3.7 and Python3.8 error

satfan52, 2019-12-09:
Hello,

Thanks for the app, but I cant' make it work in my environment. I am using Ubuntu 18.04 and I do not succeed to run the software. 

I have Python 3.6.9 (default), 3.7.5 and 3.8 installed but not much luck . The "python3.7 -m install ." and "python3.8 -m install ."  commands seems to works fine and install all the dependencies in the respective "python3.7" or "3.8 directories". If I do "python3.7 nanovna-saver.py"  or  "python3.8 nanovna-saver.py"  I have the error below. Thanks for your help

Regards
Peter

root@on7yi-ubuntu:/opt/build/nanovna-saver# python3.7 nanovna-saver.py
NanoVNASaver 0.2.1
Copyright (C) 2019 Rune B. Broberg
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Settings: /home/peter/.config/NanoVNASaver/NanoVNASaver.ini
Traceback (most recent call last):
  File "nanovna-saver.py", line 20, in <module>
    main()
  File "/opt/build/nanovna-saver/NanoVNASaver/__main__.py", line 73, in main
    window = NanoVNASaver()
  File "/opt/build/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 373, in __init__
    self.analysis_window = AnalysisWindow(self)
  File "/opt/build/nanovna-saver/NanoVNASaver/NanoVNASaver.py", line 2320, in __init__
    self.analysis_list.addItem("Low-pass filter", LowPassAnalysis(self.app))
  File "/opt/build/nanovna-saver/NanoVNASaver/Analysis.py", line 71, in __init__
    layout.addRow(QtWidgets.QLabel("Please place " + self.app.markers[0].name + " in the filter passband."))
IndexError: list index out of range
root@on7yi-ubuntu:/opt/build/nanovna-saver#

mihtjel, 2019-12-09:
Hello Peter,
I think the issue is that you have ended up setting the number of markers to 0 - making the attempt at getting the name of the first marker fail. Whoops.

If you edit your settings file (NanoVNASaver.ini, location in your error text) there's a setting in there, MarkerCount, which you can set to at least 1.

Hope this helps!

satfan52, 2019-12-09:
Hello Mihtjel,

Yes ! that does it ! 
Thanks for your help!

zarath, 2019-12-14:
Is there already a preferred Solution for this Bug?
Would it be better to force that at least one Marker is available, issue a warning if one tries to run marker depending analysis or disable them when no marker is available?
I would prefer solution one.

zarath, 2020-01-12:
Fixed in v0.2.2

closed 2020-01-12 by zarath