NanoVNA-Saver/nanovna-saver #115: Support calibration in continuous sweep mode

gabriel-tenma-white, 2019-12-09:
I have been using nanovna-saver and overall the UI is pretty well designed (and not sluggish like most things these days), but the calibration procedure is really tedious. Currently I have to stop the sweep and run through a long "wizard" and have no feedback of what my calibration data looks like. I would like to see the smith chart in realtime as I connect my cal standard so I can know whether the graph is stable or otherwise wrong (e.g. the cable is broken and the graph is jumping, or the load standard is broken and isn't near the center of the smith chart like it should be). I think something similar to the NanoVNA's calibration menu would be good, you connect a standard and click "measure (open|short|load)" (while in continuous sweep mode). The cal standards can then also be measured in whatever order the user likes. I think defaulting to continuous sweep would be nice too. 

P.S. consider renaming return loss to mag(S11) or |S11| (or reflection coefficient). That will shut up the positive vs negative return loss debate.

mihtjel, 2019-12-09:
Hi Gabriel,
do the individual solti buttons in the calibration window not do what you require?

Re: naming: There's an option to flip the return loss sign, so I consider that debate done. :-)

gabriel-tenma-white, 2019-12-09:
Seems to not work in continuous sweep mode, it always says "Two of short, open and load returned the same values". I think the code grabs the data from the last time sweep was stopped rather than what's on the screen; I can only get it to work with single sweeps. Also it complains if I try to apply calibration while continuous sweep is running.

mihtjel, 2019-12-09:
I'm certain that it grabs whatever is being displayed on the screen. (Because that's where it grabs it from :-) )

I'll test it out when I get some time to sit down and look at it. It's supposed to work at least.

Applying while sweeps are running is disabled because it caused crashes. I might look at it.

gabriel-tenma-white, 2019-12-09:
Ok, so it seems that SweepWorker in continuous sweep mode always passes the same array to app.saveData(), so app.data always points to the same array and since the CalibrationWindow only grabs a pointer to app.data when you click the open/short/load buttons it ends up with the same data for all cal standards. In single sweep mode it re-creates self.data* arrays.

mihtjel, 2019-12-09:
Good point, it should make a copy.

solhuebner, 2021-03-22:
@gabriel-tenma-white is this still an issue?

closed 2022-12-12 by github-actions[bot]