NanoVNA-Saver/nanovna-saver #121: qt.qpa.plugin

jponko, 2019-12-23:
I'm using Linux Mint 19.2 which is based on Ubuntu 18.04. The "python3.7 -m pip install ." command : "Successfully installed NanoVNASaver-0.2.1 PyQt5-5.14.0 PyQt5-sip-12.7.0 numpy-1.18.0 pyserial-3.4 scipy-1.4.1"

But when I try to run "python3.7 nanovna-saver.py" I get the following error:

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

Aborted (core dumped)
--------------------------------------------------------
I have the Qt 5.4.12 64-bit development system installed. Is the Qt 5.4.12 xcb plugin incompatible with pyQt5? BTW, I'm not a python programmer.

I ran ldd  on the libqxcb.so plugin but I didn't see any errors:
jponko@jponko-cb:~/opt/Qt/5.12.4/gcc_64/plugins/platforms$ ldd libqxcb.so
	linux-vdso.so.1 (0x00007ffd2d9f0000)
	libQt5XcbQpa.so.5 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libQt5XcbQpa.so.5 (0x00007f69ea8e9000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f69ea6a4000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f69ea3f0000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f69ea1d3000)
	libQt5Gui.so.5 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libQt5Gui.so.5 (0x00007f69e9987000)
	libQt5DBus.so.5 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libQt5DBus.so.5 (0x00007f69e96fb000)
	libQt5Core.so.5 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libQt5Core.so.5 (0x00007f69e8f62000)
	libGL.so.1 => /usr/lib/x86_64-linux-gnu/libGL.so.1 (0x00007f69e8cd6000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f69e8ab7000)
	libX11-xcb.so.1 => /usr/lib/x86_64-linux-gnu/libX11-xcb.so.1 (0x00007f69e88b5000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f69e868d000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f69e8483000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f69e8271000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f69e7f39000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f69e7b9b000)
	libxkbcommon-x11.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon-x11.so.0 (0x00007f69e7993000)
	libxkbcommon.so.0 => /usr/lib/x86_64-linux-gnu/libxkbcommon.so.0 (0x00007f69e7754000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f69e7550000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f69e71c7000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f69e6faf000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f69e6bbe000)
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f69e69bc000)
	libglib-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f69e66a5000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f69e6473000)
	libpng16.so.16 => /usr/lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f69e6241000)
	libdbus-1.so.3 => /lib/x86_64-linux-gnu/libdbus-1.so.3 (0x00007f69e5ff4000)
	libicui18n.so.56 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libicui18n.so.56 (0x00007f69e5b5b000)
	libicuuc.so.56 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libicuuc.so.56 (0x00007f69e57a3000)
	libicudata.so.56 => /home/jponko/opt/Qt/5.12.4/gcc_64/lib/libicudata.so.56 (0x00007f69e3dc0000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f69eae8e000)
	libGLX.so.0 => /usr/lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f69e3b8f000)
	libGLdispatch.so.0 => /usr/lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f69e38d9000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f69e36d5000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f69e34cf000)
	libxcb-xkb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb-xkb.so.1 (0x00007f69e32b3000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f69e3041000)
	libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f69e2dbd000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f69e2ba8000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f69e29a0000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f69e277a000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f69e255e000)
	libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f69e2243000)
	libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f69e202e000)

zarath, 2019-12-23:
Have you tried Version 5.14 of qt ...
As PIP's PyQt5  is  5.14.0 and your ldd output shows  Qt/5.12.4/

jponko, 2019-12-23:
I removed Qt5 development and did a "sudo apt install qt5dxcb-plugin" from
the command-line and it now nanonova-saver runs. I tried to install Qt
5.14.0 but it also wouldn't run because it also couldn't initialize the xcb
plugin. Now I'll try to reinstall the Qt 5.12.4 LTS development system and
see if that will run as well. I guess you can close the ticket.

FYI on qt5dxcb-plugin, although I've never heard of it myself:

Qt platform theme integration plugin for DDE
Deepin's qt5dxcb-plugin is the Qt platform integration plugin for Deepin
Desktop Environment. It provides an abstract interface between Deepin
applications and Qt tookit for further application theme customization. Qt
platform theme integration plugin for DDE

Thanks,
Jerry Ponko
AC9NM

closed 2019-12-24 by zarath