NanoVNA-Saver/nanovna-saver #122: Please add external bridge support

yo2naa, 2019-12-23:
Please add external bridge support, as described in messages:
https://groups.io/g/nanovna-users/message/7685
https://groups.io/g/nanovna-users/message/7693

Using an external bridge would enable more precise measurements for very high or very low impedance, using a different reference than 50 Ohm.

Thank you, I really appreciate all your work.

mihtjel, 2019-12-23:
Hmm, those posts aren't particularly specific about what is required?

As far as I can tell, they are by Erik, and describe how he's taken the code that's already in NanoVNA-Saver, and translated it to a different language?

Supporting a reference impedance different from 50 ohms would not be too difficult - I'll put that on the wish list for the future.

zarath, 2019-12-23:
RFTools are already prepared for non 50Ohm references ... but some datastructures has to be extended. I think in that future one could also support 4 port touchstone files....
Wishes for next christmas ;-)

yo2naa, 2019-12-23:
I'm sorry I was not very clear. Using a non 50Ohm reference would be another nice feature. 
I'm afraid calibrating NanoVNA with a non 50 Ohm reference is not ideal, internal bridge reference is fixed to 50 Ohm.
By using an external bridge, a different reference can be used. The problem is that the output of the bridge is connected to CH1:
https://groups.io/g/nanovna-users/attachment/7650/0/reflBridge.jpg
I think what is needed is a calibrated S11 measurement by CH1, instead of the usual CH0.
So for calibration, we need measurements from CH1 with Short, Open and Load connected on the DUT port of the bridge.
Than calculate S11 by CH1, and from S11 the impedance of the DUT connected to the bridge.
Sorry for my lack of knowledge and understanding, I tried to explain as best I can.

erikkaashoek, 2019-12-24:
Indeed,
The only thing needed to make this work in nanoVNA-saver is the ability to set the application in "one port external bridge" mode where the uncalibrated S11 data is taken from port2 (CH1) instead of port1 (CH0) and no S21 data is measured. Everything else stays as it is. Doing so will enable connecting an external bridge to port1 (CH0) for the signal input and to port 2 (CH1) for the bridge output. Connect any reference impedance to the bridge you like and connect the DUT to the dut port of the bridge. As the nanoVNA internal bridge is fairly good and you lose some dB in signal the noise will increase and the frequency range will also not see a big difference but the biggest difference is the ability to use none 50 ohm references allowing extension of the practical impedance measurement range.

mihtjel, 2019-12-24:
Okay, that's very doable. I'll put that on the list :-)

closed 2022-12-12 by github-actions[bot]