NanoVNA-Saver/nanovna-saver #123: Can't see display on 4K monitor

dmandala, 2019-12-28:
The entire display is unreadable on a 4K monitor when the monitor is not scaled.

Open the application on a 4K monitor.  Can't read anything, far too tiny.  Moved application onto a 1080p monitor and can see it. Went to the "Display Setup" and selected 12 (largest font size offered) and then dragged application back onto the 4K display.  Still too small.   Went into python code found where Font Size is set, added more font sizes including 24 and 36.  Saved file reran application, 24 usable but somewhat small, 36 usable but some input boxes have text overrunning  them.

Suspect that just grabbing a much larger font is not the correct fix for high res high DPI display but it at least makes it possible to use.

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.10 standard desktop on high DPI 4K screen
 - Python 3.6
 - NanoVNA-Saver version: 0.2.]

mihtjel, 2019-12-28:
Hi David,
please try running with the environment variable QT_AUTO_SCREEN_SCALE_FACTOR set to 0:

QT_AUTO_SCREEN_SCALE_FACTOR=0 ./nanovna-saver.py

If this fixes it, it points to the Qt scaling being "wrong" in your installation - or rather, different from the way scaling is working on Windows and Mac OS, where it needs to be set to 1, as I understand it.

Hope this helps!

dmandala, 2019-12-28:
With "QT_AUTO_SCREEN_SCALE_FACTOR=0" it is much much better, dropped my Font back down to 7 and I can read the display easily and it's mostly correct.  

Thank you.

There is still something minor wrong,   Some of the click boxes are still over running their buttons.  I just did a new clean DL from here and its still slightly messed up so it's nothing I did when I changed the font sizes.  I've attached a screen capture of the current display with Font at 7.
 
[image]

Thank you very much for the quick response.

closed 2020-11-01 by zarath