NanoVNA-Saver/nanovna-saver #124: Unable to compile on Raspberry Pi Raspbian Buster

jimmycapizzi, 2019-12-30:
Getting the following error...
chubb@K3CHB-Pi:~/src/nanovna-saver master ± pip3 install .
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /home/chubb/src/nanovna-saver
Collecting PyQt5 (from NanoVNASaver==0.2.1)
  Using cached https://files.pythonhosted.org/packages/7c/5b/e760ec4f868cb77cee45b4554bf15d3fe6972176e89c4e3faac941213694/PyQt5-5.14.0.tar.gz
  Installing build dependencies ... error
  Complete output from command /usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-6cl7r1cw --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- "sip >=5.0.1 <6" "PyQt-builder >=1.1.0, <2":
  Invalid requirement: 'sip >=5.0.1 <6'
  Traceback (most recent call last):
    File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/requirements.py", line 93, in __init__
      req = REQUIREMENT.parseString(requirement_string)
    File "/usr/share/python-wheels/pyparsing-2.2.0-py2.py3-none-any.whl/pyparsing.py", line 1632, in parseString
      raise exc
    File "/usr/share/python-wheels/pyparsing-2.2.0-py2.py3-none-any.whl/pyparsing.py", line 1622, in parseString
      loc, tokens = self._parse( instring, 0 )
    File "/usr/share/python-wheels/pyparsing-2.2.0-py2.py3-none-any.whl/pyparsing.py", line 1379, in _parseNoCache
      loc,tokens = self.parseImpl( instring, preloc, doActions )
    File "/usr/share/python-wheels/pyparsing-2.2.0-py2.py3-none-any.whl/pyparsing.py", line 3395, in parseImpl
      loc, exprtokens = e._parse( instring, loc, doActions )
    File "/usr/share/python-wheels/pyparsing-2.2.0-py2.py3-none-any.whl/pyparsing.py", line 1383, in _parseNoCache
      loc,tokens = self.parseImpl( instring, preloc, doActions )
    File "/usr/share/python-wheels/pyparsing-2.2.0-py2.py3-none-any.whl/pyparsing.py", line 3183, in parseImpl
      raise ParseException(instring, loc, self.errmsg, self)
  pyparsing.ParseException: Expected stringEnd (at char 12), (line:1, col:13)
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "/usr/lib/python3/dist-packages/pip/_internal/req/constructors.py", line 253, in install_req_from_line
      req = Requirement(req)
    File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/requirements.py", line 97, in __init__
      requirement_string[e.loc : e.loc + 8], e.msg
  pip._vendor.packaging.requirements.InvalidRequirement: Parse error at "'<6'": Expected stringEnd
  
  
  ----------------------------------------
Command "/usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-6cl7r1cw --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- "sip >=5.0.1 <6" "PyQt-builder >=1.1.0, <2"" failed with error code 1 in None

G4DCP, 2020-01-23:
For information I also have this problem and don't know how to proceed as I have SIP 5.1.0 installed.

Peter

mihtjel, 2020-01-24:
Hi,
these errors seem to be PyQt5 failing to install/build on your system. It's not an error in NanoVNA-Saver as such - and it's certainly not anything I have the knowledge to help with :-)

closed 2020-11-01 by zarath

WheezyE, 2021-01-09:
A while back, I was able to get another PyQt5 python/pip program, called Artemis 3, working on the Pi.  I made a [Raspbian build script](https://github.com/AresValley/Artemis/blob/master/spec_files/Linux/raspbian_build.sh) that might be able to be re-worked for NanoVNA-Saver.  I'm not sure yet if I'll have time to script or test anything this week, but thought you all might be interested.

WheezyE, 2021-01-09:
[image]
This seems to be working on Pi4 (PyQt5 wheel compile time ~2hr).  I don't have a NanoVNA to test it though.  Here's a pastebin of the script that gets all the pip modules installed on a non-System Python virtual environment: https://pastebin.com/b6j7wyfJ

It takes 2 hours to install modules on a Pi4 (and more than 12 on earlier Pi models).  I know that Artemis has the ability to make a binary file for certain OS's, including the Pi - I think using a .spec file?  Would it be possible to do that with NanoVNA Saver?

WheezyE, 2021-01-09:
I've been trying to follow [this guide](https://realpython.com/pyinstaller-python/) to make a binary (with the pyenv activated that I built from that pastebin script), since I know that Artemis makes its executable using pyinstaller.  Running these commands  gives me a dist folder with an executable and a bunch of shared object files, but I can't seem to execute them:
```BASH
(NanoVNASaver_python3.7.0pyqtnew) pi@raspberrypi:~/Downloads/nanovna039/nanovna-saver-0.3.9-pre $

pyinstaller nanovna-saver.py
pyinstaller setup.py
```

I keep getting errors though with my built executables though.  I think maybe the python code might just need a tiny bit of reconfiguring somehow to make it buildable and distributable.  I know very little programming other than some bash though

kj5wi, 2021-01-25:
Watch out for line endings. Geany or Cream can shift them for you. Had to do that for above script to run it on Raspbian Buster.

WheezyE, 2021-01-25:
Oh I didn’t know about that - thank you! I’m learning as I go 😅

kj5wi, 2021-01-25:
Me too. Just got caught by another one. The script builds in a dot folder under pi in my

kj5wi, 2021-01-25:
Just got another reminder. Seems that the script runs from a dot folder on ntspawn64 under pi. This time it might ready to run.

WheezyE, 2021-01-26:
Sorry I’m confused. Is this a problem that running dos2unix on the script would fix? I’m still a Linux noob (I’ve noticed that when I wget a pastebin script I have to run it through dos2unix before I can run it)