NanoVNA-Saver/nanovna-saver #125: Add attenuator and reference trace offset  to ease GAIN measurelent active devices  (two port measurement)

satfan52, 2020-01-01:
Hello,

I am using nano-vna saver to measure the gain of my 3 stage home made power amplifier for my redpitaya sdr based transceiver. The total gain of the 3 stages is in order of 50 dB, to amplify a 5 dBm signal (3mw)  to 55dBm (300 W). At the moment I am inderting a 60dB  attenuators in the rf path in order to measure the total gain of the 3 stage. For obvious reasons, the calibration is done without the 60dB attenuator  inserted the rf-path, but the sweep is of course done with the attenuator inserted (also for obvious reasons ;-) .  At the moment, I have to manually add 60dB to the sweep trace in order to obtain the GAIN of my 3 stage amplifier, that is ok of course but this is not very convenient. It would be much better if  there would be a possibility to manually enter an attenuation value, so the trace shown after the sweep is actually the real gain of the amplifier (no need to manually add 60 dB to it).  In addition to manually entering an attenuator value, it better would nice to have the possibility to ADD  the "reference trace" to the result of the sweep; the idea would be to first sweep the attenuator and define is as the reference trace. Then the active DUT would be sweep and the trace of the attenuator would be added to it to obtain the GAIN with a high precision. See example, below, it is the gain of the second stage measured with a 20dB attenuator in the rf path.

[image]

stdevPavelmc, 2020-02-02:
I'm on that exact scenario, in my case using a simple 20dB attenuator. Having a setting that allow us to declare a preset for the attenuator will be extemelly useful.

I can propose a fix, what about a field in the "Sweep settings..." tab, below the averaging option, something like this...
[image]

I'm working on the inside math to understand the implications of that change. @mihtjel any advice on that?

stdevPavelmc, 2020-02-02:
Ok, I now have a better understanding of the math operations inside (but not full insight yet)...

Gain is calculated inside the Datapoint data type, so what will be the better implementation for the gain adjust when we have an attenuator inline with the s21 input?

From my yet limited point of view there is at least two options for this:

**Adding a new data component for the Datapoint data type:** the attenuation, so when we get the gain we can add the att value in every calculation of the gain.

- Cons: Real values will not be saved on the files on export, so when you recall the file the att will be gone if you don't set in on the attenuator setting prior to the recall of the file. It will be counter intuitive...

**Calculate the inverse of the log with the attenuation added:** this will carry it's own math problems.

- Pros: Real values will be stored on files, so recall will work
- Cons: heavy math and not sure it will be technically correct, need to run a few iterations to check if the math does work.

**Any other option better than that**

@mihtjel any other ideas?

stdevPavelmc, 2020-02-02:
My math now seems good.

I we take the route of doing the math, adding the att level and taking back the math at least in my opinion it's fine.

The only sacrifice here is the imaginary part, the reactance of the output amplifier that will be masked by the attenuator; so if you uses a good attenuator then the imaginary part is negligible (and very very low if your att is a good one).

So with this trick we can use any (very good) att in the s21 line and get a level value directly into the plot and that value will be saved into the file and recalled back.

Hum... that poses a risk, we will need to reset the attenuator value when we load a file with s21 values or the values will be altered.

Taking finger for a dance on the keyboard... PR soon... Cheers

satfan52, 2020-02-02:
Hi Guys,

I am juping tje conversation

nMy idea 💡behind this request is that the offset would be added
mathematically to the trace of S21. Of course I also assumed that if I hit
the button "set as reference" then the whole trace with the offset would be
saved as a reference. This is very important of course when tuning an
amplifier to be able to compare effect of changes between two sweeps.

I use nano-vna saver in cimbination with SimSmith to manipulate S
parameters etc... So yes, it is important that the s2p touchstone file
exported adds the offset so that what is exported is what is measured.

If the attenuator is inserted on CH0, then the S11 trace will of course be
innevirably affected. To avoid that, I usually calibrate with a small
attenuator in the order of 5 to 20 dB on CH0 so I reduce the output power
of the Nanovna to  an adequate level for the measurement but without
preventing the measurement of S11. I do the measurement with a second more
iportant attenuator ( in the order of 30 to 50 dB) inserted between the
output of the amplifier and CH1. Of course, the trace should show the total
gain accounting for the effects of both aytenuators.

What would be ideal of course is to be able  to calibrate without any
attenuators in the rf path and have  nanovna-saver deembed S11 so the
effect of the CH0 attenuator is removed on the S11 trace.
I am afraid this will also require two attenuator fields fir CH0 and CH1 to
be able to make that calculation.

A variant, perhaps the most reliable or professional, could be to add TWO
additional steps at the end of the calibration procedure, to measure the
S11 and S21 parameters of the two CH0 and CH1 attenuators inserted before
and after the amplifier . Then the nano-vna software would actually remove
effects of those attenuators ( using their measured S21 and S11 traces)
when sweeping the amplifie.  Of course, for tjis to work accurately , the
value if each attenyator should not exceed 30dB, but that makes 60 dB total
which is sufficient for most amplifier  measurements.

satfan52, 2020-02-02:
If only one field is possible, then two attenuators can still be inserted
on CH0 or CH1 ports and the sum of the value of the two attenuators would
have to be entered in the software. This would already be quite nice!!!

It would not be possible to measure S11 in the same sweep unless of course
only one attenuator is used on CH1 or only one attenuator is used in CH0
and the S11 trace is recalculated accordingly.

satfan52, 2020-02-02:
To simplify things. I guess assuming perfect attenuators with S11 = S22=0
and with S21=S12=attenuation value is the way to go.

Regarding the pps delay introduced by the one or two attenuators inserted,
it could be compensated via the generic pps parameter or an additional
attenuator pps field could be foreseen for that specific purpose (up to the
user to know what he is doing and to ensure he uses attenuators with known
and constant attenuation value and pps delay over the frequency range of
interest for the measurement )

mihtjel, 2020-02-02:
This is quite an interesting discussion. However, I'm not sure I understand what the problem would be in calibrating within the application itself using the attenuator in the "through" part of the calibration? This should produce the desired results, and also compensate for any non-ideal attenuator characteristics.

satfan52, 2020-02-02:
Please refer to the graphics I posted, the reference trace with a 60db
attenuator inserted is just not usuable given the limited dynamic range of
the nanovna.
I woukd work witj smaller attenuation value up to 30dB though

stdevPavelmc, 2020-02-03:
> This is quite an interesting discussion. However, I'm not sure I understand what the problem would be in calibrating within the application itself using the attenuator in the "through" part of the [...]

The problem is that if the attenuation required is to high (think in a 100W power amplifier: 50 dBm) then the attenuator needed would be in the 60-70 dB to be safe.

Now when you calibrate with a 60 dB attenuator you are calibrating almost over the limit of the device, aka: you loose accuracy; if you can calibrate in a normal "trough" and add mathematically the 60 dB of attenuation then the displayed results will be right on an you will no lose accuracy.

I have a high gain LAN for 70 cm (+34 dB calculated, a monster I know) and it's very sensitive to the input power, to measure it I add a 20 dB attenuator in the CH0 path and then a 20 dB attenuator in the CH1 path, totally 40 dB of attenuation.

Here the S11 ans S22 plots will be a hard guess as the attenuators mask the reactance and I only care about the gain.

With a 40 dB attenuators in 70cm band and the noise floor of my device being about ~60 dB there are room to measure it but then the levels are dangerous for the hardware.

The output of the CH0 is about what? -13 to -9dBm... typically -10 dBm, the distortion level of my LNA is about +10 dBm.

With the input of the nanovna set to -10 dBm I will get a +24 dBm at the end of the LNA... so will distort, need an attenuator at the input 20 dBm is fine as -10 + -20 + +34 ~ 4 dBm just fine, but to be safe I placer another 20 dB attenuator in the output as the IP3 of the SA612 is -13 dBm...

As you see there are cases where setting the attenuator only in software has is use case.

Cheers.

stdevPavelmc, 2020-02-03:
> Please refer to the graphics I posted, the reference trace with a 60db attenuator inserted, 60dB is just not usuable given the limited dynamic range of the nanovna. What you suggest works with smaller attenuation value up to 30dB though, but for high attenuation values in the order of 60dB it does not.
> […](#)

Yes, exactly! 

This is what I'm talking about.

mihtjel, 2020-02-03:
Thanks, that made it more clear. I will take a look at it :-)

satfan52, 2020-02-03:
Absolutely!!! 😊
In practices when measuring amplifiers with gain, an attenuator is required
at the input of the amp to protect the amp and at the input of the nanovna
to protect it. The sum if the attenuation value should be ideally in the
order of magnitude of the expected gain if the amplifier measured. Even the
cheap 35dBm, 50dB gain amplifiers sold on ebay for 7 euros require to use
atrenuators on their input and outputs to measure them so it is not only
about measuring big powerfull amplifiers but also LNAs and everyday cheap
PA.

satfan52, 2020-02-03:
Moreover, if the attenuator placed on the input of rhe amp is less than say
20- 30dB (to be discussed) it can remain in calibration path so S11 can
still be measured with reasonable accuracy. However for the one placed at
the output of the amplifier it is another story. If its value is less than
30-40 dB (again to be discussed) then a separate dedicated calibration
could be ennvisaged to measure it. If the attenuation value is higher then
there is no choice, the value must be entered manually and mathematically
substracted.

Regarding the S21 phase shift of that second attenuators, there are several
options as discussed previously.

satfan52, 2020-02-03:
Maybe to simplify, simply let the user enter the sum of  attenuation value
and the electrical delay for both attenuators and then do the math to
recalculate S11 and the phase of S21

stdevPavelmc, 2020-02-03:
> Maybe to simplify things, simply let the user enter the attenuation value in dB and the electrical delay in pps for each of the two attenuators inserted at the input and at the output of the amplifier and then do the math to recalculate S11 and S21 assuming ideal attenuators. Calibration procedure would be untouched this way. It would be a simple math calculation to recalculate S11 and S21 assuming perfect attenuators over the frequency range of interest. You could write "Sometimes it is necessary to insert two attenuators at the input and output of an amplifier to measure accurately its gain and input impedance over the limitted dynamic range of the Nanovna and/or to protect both the amplifier and the Nanovna against possible overdrive and overload. The attenuation value at the input of the amplifier (connected to the CH0 port of the nanovna) should be chosen to avoid overdriving and overloading the amplifier as it could result in its destruction and/or could create unwanted distorsion affecting accuracy of measurements. The attenuation value at the output of the amplifier (connected to CH1 port of the nanovna) should be chosen to avoid overloading the CH1 port and possibly damaging the Nanovna. In addition, for optimal accuracy of measurements, the sum of the two attenuation values should be in the order of the "expected gain" of the amplifier (plus a small security factor). Please enter the attenuation value (in dB) and the delay (in pps) for each of the two attenuators inserted and nanovna-saver will then determine S21 and S11 parameters, the input impedance and the GAIN of the amplifier as if no attenuators had been inserted. The two attenuators inserted should behave as ideal ones over the frequency range of measurement, with constant attenuation and electrical delay and with flat 50 ohms input/output impedances. All values are set to 0 by default. Calibration should be done normally without any attenuator inserted" Field 1: Attenuation value in dB of the attenuator inserted at the input of the amplifier (ie, connected to the CH0 port of the Nanovna) Field 2: Electrical delay in pps of the attenuator inserted at the input of the amplifier (ie, connected to CH0 port of Nanovna) Field 3: Attenuation value in dB of the attenuator inserted at the output of the amplifier (ie, connected to CH1 port of Nanovna) Field 4: Electrical delay in pps of the attenuator inserted at the output of the amplifier (ie, connected CH1 of Nanovna)

In my opinion you don't need two values of att & delay, from the point of view of the device under test (DUT) it's not needed to know that there is two attenuators, just one global value is enough as it's a simple sum of levels & delays, the reactance part will be masked by the attenuator in it's path, then just delay and attenuation matters here.

It's up to the operator to know the pros & cons of each method, adding just an attenuator and a delay value is enough, both values can me accessed even with the nanoVNA beforehand.

Cheers.

satfan52, 2020-02-03:
If you enter only two values, so phase and delay for both attenuators then
you can recalculate amplitude and phase of S21. Thats already quite a nice
improvement as the gain and phase shift of the amplifier will show up right
on the S21 curve. I would be quite happy with such a change already.

Regarding the S11 reconstruction challenge with an attenuator connected on
the  CH0 port due to its damping effects, I just did a few tests in
REFLECTION mode with several attenuators conected to CH0 on one side and
terminated by a short on the other side. The value of S11 measured should
be theoretically twice the value of the attenuator tested, right? . This is
indeed the case for attenuator of 5 and 10 dB but above 10dB things get
messy and very very creative very quickly, especially above 300 mhz... So
practically this means it should be possible to reconstruct S11 with good
accuracy for an attenuation value of up 10dB on the CH0 port.

gaionim, 2020-03-11:
Hi,
can positive offset  be useful with un-un loss measure  ? 
see  https://www.rogerk.net/forum/index.php?topic=74641.msg784261#msg784261
can set resistence as input ( 400 or 390 as in sample ) and calculate offset as 10*log(50/resistence)

satfan52, 2020-03-11:
Hi there,

No sure I got the point, sorry.

But your remark made me realise something else. In fact the software should accept both a negative offset (use of attenuator) or a positive one (use of a preamp of a known and flat gain over the frequency of measurement)
- the attenuator is needed for measuring high gain and/or powerfull amplifiers 
- a good preamp, **with proper shielding**, may be usefull for measuring very high attenuation values (> 60-80 dB). Especially at  frequencies above 500 mhz where the dynamic range of the Nano VNA degrades fast compared to below 100 Mhz.

gaionim, 2020-03-11:
> No sure I got the point, sorry.
in this schema https://www.iz2uuf.net/img/T140-2_9a1_schem.jpg 50Ohm is the ch1 port

With 2400+50 in my 49:1 un-un I read -17.5 , but the real loss is -17.5+16.9 = 0.6 
10*log(50/2450) = 16.9
if I can't find 2450-50 = 2400 resistor but only 2200, I can use new offset of
10*log(50/2250) = 16.5

Anyway I hope https://github.com/mihtjel/nanovna-saver/pull/145 will be merged.

satfan52, 2020-03-11:
Hello,

Maybe you could describe in words what you are trying to measure, it will
be easier to understand, at least for me ( as my Italian has not
unfortunately improved recently, so not much use of the forum links you
gave) . If the goal is to measure the rf transformer ratio over frequency,
a simple solution is to attach a 50 ohm load on the 50 ohm port of the
transformer and then to measure the SWR on the other port. The SWR ratio
gives you the imledance transformation ratio, divide or multiply depending
on your transformer

gaionim, 2020-03-11:
I try to measure the insertion loss of the transformer.
You can measure the loss of 2 transformer back to back, so the impedence is 50Ohm with S21
or you can try the partition in the schematic, but the loss measured s21 logmag is the real loss plus the loss on the resistance of the partition.
Sorry, but my english is only little better than your italian ;-) You can try google translate
https://www.rogerk.net/forum/index.php?topic=74641.msg784261
the guy writing is far better than me also in italian

satfan52, 2020-03-11:
Hello again. I  now better understand many thanks.  If the goal is to compare the loss of two RF-transformers then I don't see why you want to absolutely use a resistor in the circuit, it can be done the way you suggest but it is not absolutely necessary and at higher frequencies  I am not so sure it is a very good idea to use high value resistors of up to 2.4 k !  The sum of the power lost in dissipation (heat)+ radiation (em waves ) and coupling (k) can be calculated from S11 and S21 using a simple mathematical ! What would make sense is to add this formula to the list of  "Displaycharts" available in the "Display" menu nanovna rather than playing around with resistors.

The two displaycharts that could be added are :

**LOSS = 10* LOG (1-S11*S11-S21*S21) 
and
EFFICIENCY = 10 %LOG (S11*S11+S21*S21)**

Explanation:

1) The power reflected by the transformer (so absorbed by the CH0 port assuming a negligible coax cable loss, is  S11*S11  (or 10* LOG (S11*S11) in dB)
2) The power transmitted to the load (so absorbed by the CH1 port assuming negligible cable loss is  S21 *S21  (or 10*LOG (S21*S21) in dB)
3) With an ideal device that does not dissipate, radiate any energy and with no coupling loss , you have the relationship,  S11*S11+S21*S21 = 1  (note that S11 and S21 are NOT expressed in dB here) as no energy is lost in the system.
4) with a non ideal device, so a real world device, the power LOSS dissipated and radiated or due to bad coupling is logically and very simply :  1 - (S11*S11 + S21*S21). So if nanovna is modified to also calculate this specific LOSS value (in dB) you have the LOSS factor = 10*log (1-S11*S11-S21*S21*)!  there is no need to play around with resistances in serie with the CH2 port to get it !!. 
5) The EFFICIENCY factor is simply, S11*S11 +S21*S21 and this is always < 1 (for passive devices). When expressed in dB it is 10*LOG (S11*S11+S21*S21) and it is always less than 0dB.  For a perfect device with no power radiated, dissipated or lost in coupling, the efficiency factor is 0 dB.  

More details are available here
https://www.microwaves101.com/encyclopedias/mismatch-loss-etc

gaionim, 2020-03-12:
>  why you want to absolutely use a resistor in the circuit, it can be done the way you suggest 

but I need 2 equal devices

> At higher frequencies I am not so sure it is a very good idea to use high value resistors of up to 2.4 k (stray capacitance) ! 

MY needs are only HF ( 80,40,20 m ) for EFHW antenna 

> The sum of the power lost in dissipation (heat)+ radiation (em waves ) and coupling (k) can be calculated from S11 and S21  ...

but with only 1 device there is great mismatc, I'm not sure I can separate power lost I need. 

> you don't know if what you measure is due to dissipated heat, radiated energy, coupled loss in the circuit and/or simply due to a far than perfect resistor !!!!

I don't care, I'm satisfy to know if WITH NEGLIGIBLE missmath loss, other  loss is <1dB , 2dB , 3 or more 
and primarily wich configuration is better , different wire, toroid, turn ( 2to14 or 3to21), etc.

Whith 40$ instrument it seems to me enough ;-)

In any case, adding db to the measure is also useful in this case, that is for those who "play" with wire antennas and UN-UN
Calculating the value from the divider resistance is NOT necessary.

satfan52, 2020-03-12:
" but with only 1 device there is great mismatc, I'm not sure I can separate power lost I need."
That is exactly why the two formula above are useful, because it is NOT required to have a VSWR of 1 as your method that does not tolerate any mismatch (that is why you need the resistor). What is required is to have a perfect 50 ohm load (CH1 port) and a perfect Generator (CH0) with input impedance of 50 ohms!! This can be considered the case after a good calibration with  two small 3-6 dB attenuators added on CH0 and CH1 ports !  

 **### The formuila calculates pwoerloss you don't need to isolate it !**

gaionim, 2020-03-12:
I'll try some measure and math ...

gaionim, 2020-03-12:
What is wrong ?
`
In [33]: s11 = -0.949732601+ 0.118614472j
In [34]: s21 = 0.275870829 -0.015866262j
In [35]: 10 * math.log10 (1- abs(s11 * s11) - abs(s21 * s21))
Out[35]: -21.20208139716754
`
Hope the loss is 1.7 dB ( with my method ) and not 21 dB

satfan52, 2020-03-12:
Ok, I have the same number than you -21.2 dB (or 0,007582), so your calculation is right!. However I am afraid that you are not making the right interpretation. 

-21.2 dB  is the fraction of the power lost in dissipation (Heat)+radiation (EM waves)+ Bad Coupling
So for instance if your input signal is 50 dBm (100W), the power that is lost in "dissipation+radiation+coupling" inside the tranformer is 50-21.2=28.8 dbm or approximately 0.66 W.....that is intuitively the right number at such low frequencies not 1.7 dB....

In other words, -0.66 dB  this is the minimum you will lose at the frequency of measurement if your load is a perfect 50 ohm load.  You can also look at the other way and say that 99.24%= 1-0,007582 of the power of the generator is transmitted to the load if it is a perfect 50 ohm load or alternatively, that the power absorbed by the load is only -0.033dB=10*LOG (1-0.007582) less than the power of the generator assuming the load has a perfect 50 ohm impedance, any other loss is due to MISMATCH LOSS. 

So if you measure an insertion loss of 1.7 dB, it means that 0.033dB of that insertion loss is due to dissipation, radiation and coupling and that the rest is simply due to some mismatch. Note that with the figures you provided the insertion loss is -0.344db=10*LOG (S21*S21) and not -1.7 dB so something is wrong somewhere, maybe you can clarify what that number of -1.7 dB represents exactly as I may have misinterpreted

gaionim, 2020-03-12:
> -21.2 dB is the fraction of the power lost in dissipation (Heat)+radiation (EM waves)+ Bad Coupling

indee, but I do not want "Bad Coupling", so I insert the resistor and match the expected impedance of 2450 ( 50* 49 )

Here is the misure with resistor 

[image]

I had added offset to graph, but not in values.
see marker indicate s21 gain -17.48 , but is 16.9 on the resistor and only 0.58 on the un-un.
The corrected graph show better when things go wrong ( at about 21MHz, same for SWR )

satfan52, 2020-03-12:
Ok  so 0.58 dB is the total insertion loss at your measuring frequency, it means that the 0.0303 dB calculated earlier is due to heat dissipation, radiation,  bad coupling and the rest is mismatch

gaionim, 2020-03-12:
last word ;-)
0.58 loss is on different un-un ( different toroid ) 
at 3MHz
`
In [61]: s11 = complex(-0.013317577, 0.088879607)

In [62]: s21 = complex(0.136007621,-0.008694281)

In [63]: 20 * math.log10(abs(s21)) + 16.9
Out[63]: -0.4110242703447753

In [64]: 20 * math.log10 (1 + abs(s11 * s11) + abs(s21 * s21))
Out[64]: 0.22845335584963325

`

gaionim, 2020-03-12:
In [66]: s11 = complex(-0.909984171, 0.322911202)

In [67]: s21 = complex(0.278294533, -0.048830516)

In [68]: 20 * math.log10 (1 - abs(s11 * s11) - abs(s21 * s21))
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-68-61ddb04b64a3> in <module>()
----> 1 20 * math.log10 (1 - abs(s11 * s11) - abs(s21 * s21))

ValueError: math domain error

satfan52, 2020-03-12:
10  * LOG (  1- ABS (S11)*ABS(S11)- ABS(S21)*ABS(S21))

gaionim, 2020-03-12:
same error
In [70]: 10 * math.log10 ( 1- abs(s11) * abs(s11)- abs(s21) * abs(s21))
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-70-b17554bf6eca> in <module>()
----> 1 10 * math.log10 ( 1- abs(s11) * abs(s11)- abs(s21) * abs(s21))

ValueError: math domain error

satfan52, 2020-03-12:
give me the values, I will calculate it

gaionim, 2020-03-12:
the values are indicated,
s11 = complex(-0.909984171, 0.322911202)

s21 = complex(0.278294533, -0.048830516)

satfan52, 2020-03-12:
The probem is that.:  ABS (S11) * ABS(S11) + ABS(S21) * ABS(S21) > 1 as for an active device, it should be less than 1, so something is wrong with the numbers.

gaionim, 2020-03-12:
may be some LITTLE calibration problem, but it seems ok to me.
R is 0.9     near to 50/49=1.02 expected
here full data

[cb.s2p.gz](https://github.com/mihtjel/nanovna-saver/files/4325013/cb.s2p.gz)

satfan52, 2020-03-12:
I will have a close look. I think the most likely reason is that the two formula I gave you only work (sorry) if the rf transformer is terminated by 50 ohms (to be confirmed) and not by 2450 ohm.

gaionim, 2020-03-12:
ok.

satfan52, 2020-03-12:
Ok, I did several verifications

1) Regarding the formula I gave you they indeed assume a perfect 50 ohm impedance load (CH1 port) and a perfect 50 input impedance generator (CH1) port but that is not the source of the error encountered (see next point). If you put a 2.4 k resistor in serie with the CH1 port, you can still use the formula but you will have to bear in mind that they apply to the combo "transformer+resistor" as opposed to "transformer" alone.  That is the correction that you do as a matter of fact (ie 50/2450, so +16.9 dB)
 
2) I have checked your data, I have manually re-calculated everything and verified the results are the same for the SWR, the insertion Loss, the return loss than the data displayed by nanavno. I then calculated the LOSS and EFFICIENCY factor and the data the results are perfectly ok from 5.7 to 30 Mhz. Between 3 and 5.7 Mhz the LOSS is so small and the EFFICIENCY so high that it is impossible to calculate accurately those two parameters.  At 5.7 Mhz, the EFFICIENCY is -0.1 dB or 99.9% and the LOSS -29.8 dB -0.00X% . Below 5.7 Mhz, there are errors as the efficiency exceeds 100% and the percentage LOSS become positive ! Numbers are really to low

3) I have uploaded your data in Simsmith and discovered that below 5.7 Mhz the power absorbed by the Load  (CH1 port) is just a little bit bigger than the power absorbed by the transformer !!!! That is impossible of course but it shows that the source of the error is the data that is not accurate enough to calculate such LOW lossses and HIGH efficiency below 5.7 Mhz !!! 

4) Using SIM Smith, I tried tp simulate what would be the insertion loss with a LOAD of 2.45k instead of 50 ohm. In principle Simsmith allow to do that easily. The least I can say is that the result of the simulation is not encouraging at all. The best insertion loss is obtained around 19 mhz with an insertion loss of only 5 dB and a swr close to 1 but above and below it is quite a disaster. It does not match at all the results on the pictures that you posted. Try to do a sweep with a 2.4 k resistor inserted on the CH1 port, I am curious to compare with the results of the simulation.

[image]
[image]
[image]
[image]
[image]

gaionim, 2020-03-13:
I'm sorry, I didn't think I'd give you all this work.
for 1,2,3 I think I understand that these are physiological errors due to non-perfect calibration and small tolerances.
For 4 ... I do not understand, but here the full data with 2k4 resistor to match 
[bianco.s2p.gz](https://github.com/mihtjel/nanovna-saver/files/4328703/bianco.s2p.gz)

gaionim, 2020-03-13:
I took the changes from https://github.com/mihtjel/nanovna-saver/pull/145
i changed calculation to
`
def corrAttData(data: Datapoint, att: float):                                   
    """Correct the ratio for a given attenuation on s21 input"""

    if att <= 0:
        return data
    else:
        att = 10**(att/20)

    ndata = []
    for i in range(len(data)):
        freq, re, im = data[i]
        orig = complex(re, im)
        corrected = orig * att
        ndata.append(Datapoint(freq, corrected.real, corrected.imag))

    return ndata

`
to preserve phase and I'm happy with this result
[image]

satfan52, 2020-03-13:
> I'm sorry, I didn't think I'd give you all this work.
no problem, thanks to Corona-virus, my agenda is freed
> for 1,2,3 I think I understand that these are physiological errors due to non-perfect calibration and small tolerances.
yes
> For 4 ... I do not understand, but here the full data with 2k4 resistor to match
Forget it, it does not work. I tried to simulate based on the S parameter collected with a 50 ohm load, what would be the insertion loss and other parameters with a load of 2.45k. It's possible but the modeling is much more complex
> [bianco.s2p.gz](https://github.com/mihtjel/nanovna-saver/files/4328703/bianco.s2p.gz)
Thanks, I have redone the manual calculations manually and the LOSS due to radiation, dissipation, coupling is much higher now;
[manual calculation  of LOSS and EFFICIENCY V 2.zip](https://github.com/mihtjel/nanovna-saver/files/4330373/manual.calculation.of.LOSS.and.EFFICIENCY.V.2.zip)
 the EFFICIENCY much lower. About 10% of the energy of transmitter is lost in heating, radiation , coupling, etc... Of course I took into account the effect of the 2.4 k resistor in the calculation..

gaionim, 2020-03-13:
it's ok, 10% is negligible ;-)
I am a missed engineer HI

satfan52, 2020-03-13:
Welcome to the club

gaionim, 2020-03-14:
> I you pump 1KW in this baby, 100W on average will be lost and if you are a big gun operating a 2kw, it is 200W, the transformer better been build to handle that load.

my legal power is 500w and I'm little gun, only 5w ;-)

satfan52, 2020-04-22:
Can someone explain me the steps to test this feature, I have downloaded the latest version 0.22 and 0.23 and it does not appear yet. Has it been merged ? if not is there a way (patch ?) to test the enhancement proposed ?

zarath, 2020-06-27:
implemented in v0.3.0

closed 2020-06-27 by zarath

satfan52, 2020-06-29:
A Huge thanks for having implemented this change