NanoVNA-Saver/nanovna-saver #127: Smith Chart Labels

acug01, 2020-01-03:
Useful addition would be to put major division labels on the Smith Chart, e.g., normalized integers on real axis, normalized decimals on outer imaginary.  Perhaps also, additional divisions, such as on the legacy printed charts?  Thanks!

mihtjel, 2020-01-12:
More could certainly be done with the Smith chart!

satfan52, 2020-01-13:
It would be nice as well to have some admittance circles drawn on the display not only the impedance ones. It is as convenient to visualize the admittance circle for Y=1 then the circle Z=1, both should be ideally displayed. Maybe displaying the admittance circles could be another option to set in the parameters

closed 2022-12-09 by github-actions[bot]