NanoVNA-Saver/nanovna-saver #128: NanoVNA should stop sweeping if nanovna-saver has not requested it

satfan52, 2020-01-09:
It does not matter to be continuously be sweeping when measuring filters or passive devices. However when measuring amplifiers or active devices, especially with a high gain and/or high output power;  stopping the sweep in nanonovna-saver should also physically stop the sweep and the transmission of rf on the Nanovna itself.  At the moment, I have to physically disconnect the NanoVNA from the amplifier tested in order to prevent it from amplifying the signals that keep coming from the NanoVNA even if nanovna-saver has not instructed the NanoVna to sweep or alternatively,  I have to disable the tested amplifier which is not nice. This is not very convenient and it would be much better that the NanoVna simply stops sweeping when nanovna-saver does not request it.

mihtjel, 2020-01-12:
Hmm. I think for most uses, users expect the device to continue sweeping. But I get your point - having it as a setting could be useful.

satfan52, 2020-01-12:
Well, a parametervwould do it of course. I think anyone measuring
amplifiers would appreciate thst possibility

lordmorgul, 2020-01-12:
Yes most users would expect an option to trigger “single” sweep or “continuous” sweep, as is the case on most commercial network analyzers.

Andrew Farris <lordmorgul@gmail.com>
http://www.linkedin.com/in/andrewfarris
www.gideons.org
www.friendsofgideons.org

goldstar611, 2020-04-21:
I also would be interested in "single sweep" preference. I know the option is called "Pause Stimulus" on the device itself but wasn't able to find anything like that in the code base. If someone knows the serial port command to pause the sweep / stimulus I would try my hand at a PR (I'm familar with PyQt already)

goldstar611, 2020-04-21:
And the pull request is up! https://github.com/mihtjel/nanovna-saver/pull/173 Feedback would be appreciated. I just added a new checkbox in the "Sweep Settings" window that issues the `pause` command after a sweep/scan is complete.

satfan52, 2020-04-22:
Hello,
Thank you for the change. I am not really into Python, I am more of an RF guy. Can you please explain me briefly (key steps) how I can test your change ? I guess I have to apply some patch in V0.22 or is it included already in V0.23 that I should just re-donwload ?  Sorry for the naive question.
Kind Regards
Peter

goldstar611, 2020-04-22:
> Thank you for the change. I am not really into Python, I am more of an RF guy. Can you please explain me briefly (key steps) how I can test your change ? I guess I have to apply some patch in V0.22 [...]

Hi Peter, No worries! I assume that you are on Windows so I generated an EXE with the pause stimulus feature for you and anyone who wishes to test. It is available at https://github.com/goldstar611/nanovna-saver/releases/tag/v0.2.2-1 

Remember that this is just for testing and to be careful with high power devices that you mentioned like the PAs you are working with. I *think* I found all of the places where sweep is called but I may have missed some spots (I only spent 2 hours on it and there's a lot of code here) so proceed with caution!

Note: I don't know exactly how the exe files are created here since there was no Makefile so I had to come up with my own method (using pyinstaller)

satfan52, 2020-04-22:
Thanks! I am on Ubuntu 18.04 ;-) (yes I know I am full of contradictions). If you can please send me the link to a Linux version as well. Many thanks

goldstar611, 2020-04-22:
> Thanks! I am on Ubuntu 18.04 ;-) (yes I know I am full of contradictions). If you can please send me the link to a Linux version as well. Many thanks

Oh that should be even easier! Do what you normally do but with this source zip: https://github.com/goldstar611/nanovna-saver/archive/auto_pause_stimulus.zip

satfan52, 2020-04-24:
Hello,
Thank you so much for giving attention to this change request. I have only tested the linux link you sent and unfortunately  there is no change compared to the 0.22 version that I am using normally. In the two picture below you can see that the nanovna is continuously sweeping between 50 and 60 Mhz (my start and stop parameter') at a power level of about -40 dBm, there is no difference with my normal 0.22 version. It is also the same power level when I am sweeping. Are you sure you sent me the right link ;-) 

[image]
[image]

Kind Regards
Peter

goldstar611, 2020-04-24:
> unfortunately there is no change compared to the 0.22 version that I am using normally. In the two picture below you can see that the nanovna is continuously sweeping between 50 and 60 Mhz (my start and stop parameter') at a power level of about -40 dBm

Did you check the sweep settings, there is a new checkbox to enable the feature (It is not default because I don't know if that is desired by the author of nanovna-saver)
[image]

goldstar611, 2020-04-24:
> Hello,
> Thank you so much for giving attention to this change request. I have only tested the linux link you sent and unfortunately there is no change compared to the 0.22 version that I am using [...]

I hooked up a scope with a SMA to BNC adapter and indeed there is still a stimulus on the output with my changes -- but it is not sweeping anymore (pause not stop stimulus). I will dig into the firmware to see if it's possible to completely stop the stimulus but I have a bad feeling about that feature.

satfan52, 2020-04-24:
Thanks ;-) No I had not... I redid the test with the check box enabled and indeed  the screen of my spectrum analyzer is much less crowed (see picture below). However there is a huge spike of -17 dBm corresponding to the "start frequency".....that keeps "booming" after after the sweep. If you could manage of getting rid of it, it would be fantastic !

goldstar611, 2020-05-02:
So I took a look at the supported commands in the firmware and have bad news.
1. I don't know how to eliminate or reduce the start frequency signal after the sweep is finished :(
2. Firmware support to completely turn off stimulus is not supported. (at least I could not find a serial command to do it)

There is a "power" command which changes the output power of the stimulus but it doesn't disable the stimulus in my testing, only adjusting the gain.

satfan52, 2020-05-24:
Thank you so much for looking at this.  Let's keep the ticket open until someone figures out the issue. Maybe that a walk-around could be to decrease the gain to the lowest possible value until the next sweep. If we can lower the residual output signal by -40dB so from -17 dbM to -57 dBm or so it is ok too. 
Kind Regards
Peter

satfan52, 2020-07-18:
Actually the trick could be to let the nanovna sweep but disable the rf output. Some active components should be shut down via the firmware or a trick is needed to disable the rf output some ways so no rf energy exits through CH0

goldstar611, 2020-07-20:
> let the nanovna sweep but disable the rf output

I'm not sure if that's feasible. Someone please correct me if I am wrong and I will mark this comment as outdated.

I took a very very quick glance at what I think is the most popular schematic:
https://github.com/ttrftech/NanoVNA/blob/master/doc/nanovna-sch.pdf
I'm pretty sure X1 and X2 are the SMA connectors. These are directly connected to SA612AD chips in the schematic.
According to [NXP](https://www.nxp.com/docs/en/data-sheet/SA612A.pdf) it doesn't appear that this mixer has any RF OUT enable/disable mechanism.

satfan52, 2020-07-31:
I have opened a ticket to modify the firmware of the nanovna V2 

https://github.com/nanovna/NanoVNA-V2-firmware/issues/28

The idea is no longer to stop the ongoing sweep as this could have some performance impact, but to have an option in the firmware to cut the rf output. This option could then be exploited by nanovna-saver

BenjiWiebe, 2022-09-29:
> I'm not sure if that's feasible. Someone please correct me if I am wrong and I will mark this comment as outdated.
> ...
> These are directly connected to SA612AD chips in the schematic. According to [NXP](https://www.nxp.com/docs/en/data-sheet/SA612A.pdf) it doesn't appear that this mixer has any RF OUT enable/disable mechanism.

As far as I can tell from reading your linked schematic and relevant datasheets -
The ports are connected to the inputs of the SA612 mixers. The RF output is provided by the Si5351A, which is controlled by I2C and does have an output disable option.

closed 2023-01-27 by github-actions[bot]

rjordans, 2023-02-12:
The NanoVNA I have indeed does reproduce.

My LiteVNA doesn't seem to suffer from this issue, I see no RF when no sweep is running in NanoVNA-saver. 

Sounds like a hardware/firmware issue in the NanoVNA to me