NanoVNA-Saver/nanovna-saver #129: Install problem in Ubuntu 18.04

sprakashb, 2020-01-09:
The following error is shown
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-d4af0ajw/PyQt5/

when 
python3.7 -m pip install . 
is executed.

jeffkarney, 2020-01-11:
The issue seems to be with PyQt5. ```pip3 install PyQt5``` results in the same error message.

```pip3 install PyQt5==5.14.0``` seems to work.

You can force that version by modifying the setup.py file. Update the install_requires var to...

```python
    install_requires=[
        'pyserial',
        'PyQt5==5.14.0',
        'numpy',
        'scipy'
    ],
```

sprakashb, 2020-01-25:
Thanks a lot jeffkarney. It solved the issue

petem001, 2020-01-25:
Maybe it would be a good thing to have this in the main project till there is a fix in linux? Can someone running windows test it to see that it does not break anything?

sprakashb, 2020-01-26:
This worked fine and i am able to install nano vna saver successfully. Yet
to test thoroughly.

Thank you for your kind support.

Vu2spf
Bhatnagar

sesykes71, 2020-02-08:
I made the change in setup.py as suggested. No change, same error. Any ideas?
Steve KD2OM
WARNING: The directory '/home/steve/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Processing /home/steve/nanovna-saver
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-vxl52rh6/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-vxl52rh6/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-req-build-vxl52rh6/pip-egg-info
         cwd: /tmp/pip-req-build-vxl52rh6/
    Complete output (1 lines):
    You need at least Python 3.7 for this application!
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

zarath, 2020-06-27:
Should be fixed by #129

closed 2020-06-27 by zarath