NanoVNA-Saver/nanovna-saver #132: 0.2.2 Crash after loading calibration file

acug01, 2020-01-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Perform a calibration in NVSaver.
2. Save calibration file.
3. Exit and restart NVSaver.
4. Load calibration file.
5. Set Continuous Sweep.
6. Start sweep.
7. NVSaver crashes.

Crash report:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	nanovna-saver-v0.2.2.exe
  Application Version:	0.0.0.0
  Application Timestamp:	5d24a31b
  Fault Module Name:	ucrtbase.DLL
  Fault Module Version:	10.0.10137.0
  Fault Module Timestamp:	556ebf74
  Exception Code:	40000015
  Exception Offset:	000764fa
  OS Version:	6.1.7601.2.1.0.768.3
  Locale ID:	1033
  Additional Information 1:	ff52
  Additional Information 2:	ff52df55cf70ef1ee9a6fa8528cd88a6
  Additional Information 3:	f26d
  Additional Information 4:	f26d3b73534facc55216d73a307ca16d

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Python version: [if applicable]
 - NanoVNA-Saver version: [e.g. 0.1.4]

**Additional context**
Add any other context about the problem here.
> Windows 7 SP1.

[S21.cal.txt](https://github.com/mihtjel/nanovna-saver/files/4051328/S21.cal.txt)

~agc

MintoWitteveen, 2020-01-16:
Can confirm this. Here however the application just exits, without any waring or message...

mihtjel, 2020-01-17:
There is a 0.2.2-1 .exe in the 0.2.2 release now, which should fix it.

MintoWitteveen, 2020-01-17:
Hi mihtjel thanks that fixed it for me!

rmalsbury50, 2020-01-21:
I am running 0.2.2 and have crash when hitting the save option. Running Win10

mihtjel, 2020-01-22:
Looks like this issue is fixed in 0.2.2-1.

closed 2020-01-22 by mihtjel