NanoVNA-Saver/nanovna-saver #133: Crystal parameters calculation/display

vadim-dem, 2020-01-14:
Do you think adding crystal parameters calculation from their scans would be a good feature? The attached a PDF printout of a simple curve-fitting algorithm implemented in Mathcad, and zipped Mathcad file. It fits a circle into admittance B/G plot (Y=G+jB) and a line into tan(arg(Y)), thus allowing a good accuracy at reasonable frequency step of scans. Actually the fitting starts at page 7 of the printout, two other readings are also listed there: 4,782,281 US patent, http://patft.uspto.gov/netacgi/nph-Parser?Sect1=PTO1&Sect2=HITOFF&p=1&u=/netahtml/PTO/srchnum.html&r=1&f=G&l=50&d=PALL&s1=4782281.PN. and the best description of the circle fitting algorithm by Randy Bullock, https://dtcenter.org/met/users/docs/write_ups/circle_fit.pdf . All the scans used are on Google drive https://drive.google.com/drive/folders/1GLG88LrIcKK4O_8i7OJdPFRQF1mSgs48?usp=sharing and I would be happy to answer your questions 

With best regards,

Vadim
[msmt_hc49u_4m9152_07.zip](https://github.com/mihtjel/nanovna-saver/files/4061861/msmt_hc49u_4m9152_07.zip)

[msmt_hc49u_4m9152_07.pdf](https://github.com/mihtjel/nanovna-saver/files/4061793/msmt_hc49u_4m9152_07.pdf)

mihtjel, 2020-01-17:
It's certainly an interesting suggestion! I will tag it as a requested enhancement :-)

paulh002, 2020-04-02:
I do this by using the bandpassfilter analysis. The use excel to calculate Rs and Cm. You could simple enhance the bpf analysis option.

vadim-dem, 2020-04-04:
Yes, I agree completely, you can start with 3dB bandwidth or with +/- 45 degree phase points. The problem is in getting these points at reasonable accuracy. Here the circle fitting method works well even when you use a single scan of 101 points covering both serial and parallel resonances of a crystal, then besides Rs and Lm (Cm) you can also get C0. And you can use reflection measurement as well, just converting its result to G+jB first

closed 2022-11-10 by github-actions[bot]