NanoVNA-Saver/nanovna-saver #134: Connecting to NanoVNA ends in an unresponsive state of nanovna-saver.

mklengel, 2020-01-15:
**Describe the bug**
After the installation (python3 -m pip .) everything works up to the point when I try to start the connection between the App and the NanoVNA-H. Then nanvna-saver doesn’t respond and I have to kill it. The device file seems correct (/dev/cu.usbmodem4001).

**To Reproduce**
Steps to reproduce the behavior:
1. Terminal: NanoVNASaver
2. Click on: Connect to NanoVNA

**Expected behavior**
I expect a working connection.

**Screenshots**
[image]
**Desktop (please complete the following information):**
 - OS: macOS 10.14.6
 - Python version: 3.7.6
 - NanoVNA-Saver version: 0.2.2

**Additional context**
To kill NanoVNASaver I have to type Ctrl-C in the terminal, which gives the following output:

MacPro-MK:nanovna-saver mklengel$ ls -l /dev/cu.*
crw-rw-rw-  1 root  wheel   18,   5 15 Jan 11:06 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   9 15 Jan 11:06 /dev/cu.LGSOUNDPLATE4B54-BTSPP2
crw-rw-rw-  1 root  wheel   18,   1 15 Jan 11:06 /dev/cu.SLAB_USBtoUART
crw-rw-rw-  1 root  wheel   18,   3 15 Jan 11:06 /dev/cu.SLAB_USBtoUART1
crw-rw-rw-  1 root  wheel   18,   7 15 Jan 11:06 /dev/cu.iPhone8-mk-WirelessiAPv2
crw-rw-rw-  1 root  wheel   18,  15 15 Jan 19:45 /dev/cu.usbmodem4001

MacPro-MK:nanovna-saver mklengel$ NanoVNASaver
NanoVNASaver 0.2.2
Copyright (C) 2019 Rune B. Broberg
This program comes with ABSOLUTELY NO WARRANTY
This program is licensed under the GNU General Public License version 3

See https://github.com/mihtjel/nanovna-saver for further details
Settings: /Users/mklengel/.config/NanoVNASaver/NanoVNASaver.ini
^CTraceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/NanoVNASaver/NanoVNASaver.py", line 637, in serialButtonClick
    self.startSerial()
  File "/usr/local/lib/python3.7/site-packages/NanoVNASaver/NanoVNASaver.py", line 656, in startSerial
    self.vna = VNA.getVNA(self, self.serial)
  File "/usr/local/lib/python3.7/site-packages/NanoVNASaver/Hardware.py", line 49, in getVNA
    return NanoVNA_H(app, serial_port)
  File "/usr/local/lib/python3.7/site-packages/NanoVNASaver/Hardware.py", line 235, in __init__
    self.version = Version(self.readVersion())
  File "/usr/local/lib/python3.7/site-packages/NanoVNASaver/Hardware.py", line 338, in readVersion
    data = self.serial.readline().decode('ascii')
  File "/usr/local/lib/python3.7/site-packages/serial/serialposix.py", line 483, in read
    ready, _, _ = select.select([self.fd, self.pipe_abort_read_r], [], [], timeout.time_left())
KeyboardInterrupt
Abort trap: 6

MacPro-MK:~ mklengel$ head /usr/local/bin/NanoVNASaver
#!/usr/local/opt/python/bin/python3.7
# -*- coding: utf-8 -*-
import re
import sys
from NanoVNASaver.__main__ import main
if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])
    sys.exit(main())
MacPro-MK:~ mklengel$

mihtjel, 2020-01-16:
Hi, it looks like the NanoVNA is detected (I assume the port name is autodetected?) and that the connection does open - and that the device does then not respond. To me, this looks like a defective device or defective firmware. From the trace, you can see that the connection is stuck in "self.serial.readline()", which is a system call to read bytes from the device.

Have you had your NanoVNA working with any other software? Have you tried connecting directly to the serial port interface?

mklengel, 2020-01-16:
Hello mihtjel,
I don't like Windows that much but I gave it a try. The windows version works, the macOS version doesn’t. I flashed the same firmware with dfu-util on macOS, the problem on macOS remained. During the "Non-Source" flash procedure I was not sure which file I had to use. I decided to give the .bin file a try just because I've found the same one in the "build from source" documentation.

I used the precompiled firmware files and dfu-util gave me a message:
```
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
```
but the procedure continued successfully.

```
MacPro-MK:Firmware mklengel$ ls
NanoVNA-H_20191125.bin  NanoVNA-H_20191125.dfu  NanoVNA-H_20191125.hex
MacPro-MK:Firmware mklengel$
MacPro-MK:Firmware mklengel$ dfu-util --list --device 0x83:df11
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=2, cfg=1, intf=0, path="93-1", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 e", serial="FFFFFFFEFFFF"
Found DFU: [0483:df11] ver=2200, devnum=2, cfg=1, intf=0, path="93-1", alt=0, name="@Internal Flash  /0x08000000/064*0002Kg", serial="FFFFFFFEFFFF"
MacPro-MK:Firmware mklengel$
MacPro-MK:Firmware mklengel$
MacPro-MK:Firmware mklengel$ dfu-util -d 0483:df11 -a 0 -s 0x08000000:leave -D NanoVNA-H_20191125.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 96488
Download    [=========================] 100%        96488 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
MacPro-MK:Firmware mklengel$
```
Did I use the wrong file?

Building from source according to https://github.com/ttrftech/NanoVNA was my last action and it doesn't work either on macOS.
```
MacPro-MK:NanoVNA-H mklengel$ git clone https://github.com/ttrftech/NanoVNA.git
Klone nach 'NanoVNA' ...
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 1366 (delta 19), reused 25 (delta 11), pack-reused 1328
Empfange Objekte: 100% (1366/1366), 1.24 MiB | 2.79 MiB/s, Fertig.
Löse Unterschiede auf: 100% (955/955), Fertig.
MacPro-MK:NanoVNA-H mklengel$ cd NanoVNA/
MacPro-MK:NanoVNA mklengel$ git submodule update --init --recursive
Submodul 'ChibiOS' (https://github.com/edy555/ChibiOS.git) für Pfad 'ChibiOS' in die Konfiguration eingetragen.
[... 208 lines cut ...]
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 97612
Download    [=========================] 100%        97612 bytes
Download done.
File downloaded successfully
Transitioning to dfuMANIFEST state
```

Do you see any chance to solve the problem with macOS?

Regards
Michael

mklengel, 2020-01-16:
In the meantime I've found a workaround using an USB hub, which uses this kernel extension: SiLabsUSBDriver CP210x USB to UART Bridge VCP Driver 5.2.4

See the screenshots of the working configuration (NanoVNA attached to the USB hub):
[image]
[image]

and the IOJones screenshot of the non-working configuration (NanoVNA attached to the native USB port of my MacPro 5,1):
[image]

I have no explanation but the problem seems to be the communication to the native USB ports/drivers/extension(s).

Regards
Michael

mihtjel, 2020-01-17:
Hmm, this is a very odd problem - but it does not sound related to NanoVNA-Saver as much as to the operating system/device/hardware. I'm happy that you found a solution, and I hope you enjoy using the software!

closed 2020-01-17 by mihtjel

mklengel, 2020-01-17:
Hello mihtjel,

I can live with my solution. May be its useful to mention another great open source app, which works on the native USB ports of the MacPro 5,1 too:

https://gitlab.com/DavidGriffith/minipro

Regards
Michael

mihtjel, 2020-01-17:
I know that NanoVNA-Saver works with the native ports on macs from other users - exactly what's the issue on your particular setup is difficult to diagnose remotely :-)

mklengel, 2020-01-17:
Nothing special - its a standard macOS Mojave installation.
Is there a flag for debugging purposes?

mihtjel, 2020-01-17:
Yes, -d gives you debugging output, -D file.txt to save it to a file.

mklengel, 2020-01-17:
The NanoVNA works also on a PCIe based USB Host-Controller.
Has anyone working the NanoVNA on a native USB port of a MacPro 4,1 or 5.1?

mihtjel, 2020-01-17:
I don't know - sounds like a question for the nanovna-users mailing list :-)

mklengel, 2020-01-17:
Thx for the hint :-)