NanoVNA-Saver/nanovna-saver #136: Discontinuities(spikes) in measurements when > 1 segments sweep

MintoWitteveen, 2020-01-16:
**Describe the bug**
When I do a sweep with more than 1 segment there are often (large) spikes in the graphs, suggesting some sort of discontinuity / wrong measurements on the boundaries of the segments.
These are not completely reproduceable (i.e. not the same every scan). But the are absent when scanning one (1) segment EXCEPT sometimes there is a spike at the beginning or the end of the scan. See attached screenshots.
These spike also appear when I do a calibration with more than one segment (as one does ofcourse)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'sweep control'
2. Click on 'segments and put in a value other than 1 (say 10 for optimal effect)'
3. Click on Sweep
4. See effect. Attached two screenshot, one with 10 segments and one with 1 segmen

**Expected behavior**
No spices on segment boundaries :-)

**Screenshots**
Done.

**Desktop (please complete the following information):**
 - OS: W10
 - Python version: [if applicable]
 - NanoVNA-Saver version:  0.2.2

[image]
[image]

mihtjel, 2020-01-16:
What firmware version are you running? This looks a lot like a bug that used to be present in some old firmwares.

MintoWitteveen, 2020-01-16:
I am running v0.2.3-2  18 oct 2019.
Just got the nanoVNA, will check for newer firmware.

MintoWitteveen, 2020-01-16:
Hi that was it!
I upgraded from the 2019.10.18 firmware to the latest 2019.11.25 and now the spikes are gone!

Thanks for the quick response!

mihtjel, 2020-01-17:
Sounds good! Glad it's working for you, and I hope you enjoy the software.

closed 2020-01-17 by mihtjel