NanoVNA-Saver/nanovna-saver #141: S-Parameter file (Touchstone v1.1) with format "DB" are not displayed correctly

bvernoux, 2020-01-24:
**Describe the bug**
When we import S-Parameter file (Touchstone v1.1) with format "DB" (dB-angle) it is not displayed correctly.
Note:  S-Parameter file / Touchstone v1.1 with format "RI" (Real-imaginary) & "MA" (Magnitude-angle) work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Open [NanoVNA Saver 0.2.2-1](https://github.com/mihtjel/nanovna-saver/releases/tag/v0.2.2)
2. Click on 'Files ...'
3. Click on 'Load as sweep' or 'Load as reference' (issue is the same)
4. See **Screenshot/Files** with working file (attenuator-0643_RI_Hz_OK.s2p) and non working file (attenuator-0643_DB_Hz_KO.s2p) 
Those files contains exactly same data in 2 different format (RI & DB).

**Expected behavior**
The S-Parameter (Touchstone v1.1) file format "DB" shall be displayed correctly like for other format  "RI" & "MA"  

**Screenshot/Files**
In this example I have loaded following files:
* attenuator-0643_RI_Hz_OK.s2p (in brown) as sweep which is OK
* attenuator-0643_DB_Hz_KO.s2p (in blue) as reference which is KO as it shall be identical to sweep/attenuator-0643_RI_Hz_OK.s2p
[image]

See file archive to reproduce the issue: [attenuator-0643_DB_Hz_KO_attenuator-0643_RI_Hz_OK.zip](https://github.com/mihtjel/nanovna-saver/files/4109906/attenuator-0643_DB_Hz_KO_attenuator-0643_RI_Hz_OK.zip)

**Desktop:**
 - OS: Windows7 Pro 64bits
 - Python version: N/A (Use the one embedded in nanovna-saver-v0.2.2-1.exe)
 - NanoVNA-Saver version: nanovna-saver-v0.2.2-1.exe

bvernoux, 2020-01-24:
This issue should be fixed by https://github.com/mihtjel/nanovna-saver/pull/140 to be confirmed

mihtjel, 2020-01-25:
I am not sure how the pull request would be related to this? But I will take a look at your examples! :-)

bvernoux, 2020-01-26:
If you want more files for test I can provide more and I can also provide my tool which convert SnP to any format RI, DB, MA (so far I have not released it on github)

zarath, 2020-01-26:
@bvernoux thanks for finding the bug ans supplying that test data.

Can you supply the same data as ma to me, so that i can use it as test data for comparision?

Sorry, that I wasn't able to look earlier on this issue

bvernoux, 2020-01-26:
See same data in MA format 
[attenuator-0643_MA_Hz_OK.zip](https://github.com/mihtjel/nanovna-saver/files/4114242/attenuator-0643_MA_Hz_OK.zip)

mihtjel, 2020-01-27:
Thanks for fixing this, @zarath :-)

closed 2020-01-27 by mihtjel

bvernoux, 2020-01-27:
Very nice job and fast fix thanks
I confirm all work fine now

PR #143 (2020-01-27): fixed db conversation bug
Fixes #141
Accidently used e^x instead of 10^x in db conversation.
Added testcases for comparison in test_touchstone.py

Sorry for inconvenience and thanks for supplying test data.